Joshua-Riek/ubuntu-rockchip #429: No HDMI sound devices in Gnome Sound settings

ndrewpj, 2023-10-27:
Hi there! I got no sound devices for HDMI output in Gnome's sound settings (I got a monitor with speakers), but in the console I see them. What could I do with this?

Joshua-Riek, 2023-10-27:
Eh what board?

ndrewpj, 2023-10-27:
sorry, forgot it - Orangepi 5 plus, latest release of your OS

Joshua-Riek, 2023-10-28:
I just tested the latest release, and HDMI audio works fine on the Orange Pi 5 Plus. Maybe you installed a package that causes audio conflicts.

closed 2023-10-28 by Joshua-Riek

JFLim1, 2023-10-29:
> I just tested the latest release, and HDMI audio works fine on the Orange Pi 5 Plus.

Hi @Joshua-Riek,

Everytime the Orange Pi 5 Plus is rebooted Sound Output device is reset to "Speakers - ES8388 Audio" eventhough it was set to "Analog Output - HDMI0 Audio" using the "Gnome Settings".  Somehow Sound settings will reset to "Speakers - ES8388 Audio" when you reboot.

bagong, 2023-10-29:
Audio works, but as JFLim1 described, doesn't stick when set to hdmi0 or 1.

bagong, 2023-11-03:
Since I switched to pipewire for audio as well (following https://gist.github.com/the-spyke/2de98b22ff4f978ebf0650c90e82027e?permalink_comment_id=3976215), the audio device sticks through reboot. Now however the names - audio analog, hdmi0 and hdmi1 - are not distinguished, all three outputs are called "Analog output - built in Audio". One can however switch properly, and as I said, it sticks.