Joshua-Riek/ubuntu-rockchip #431: WiFi module doesn't detect any WiFi networks

lemur117, 2023-10-28:
I tried using Realtek RTL8852BE-CG in my Orange Pi 5 Plus.  The module works in the Official Debian image (KDE variant).  However in this Ubuntu installation, the RealTek device is detected but it does not detect any WiFi networks.  If I type in the SSID of my WiFi network, it also is not able to connect.

Also when I enable Bluetooth it is not visible to other devices.

I'm happy to provide any details that will help.

Joshua-Riek, 2023-10-28:
Can you put dmesg into a text file and upload it here?

lemur117, 2023-10-28:
[ubuntu_dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13196479/ubuntu_dmesg.txt)

It has this:

[    9.716061] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[    9.716066] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg

Joshua-Riek, 2023-10-28:
I just pushed a firmware and kernel update to fix Bluetooth on the rtl8852b. Is this the official Orange Pi 5 Plus WiFi + Bluetooth module?

lemur117, 2023-10-28:
I think it's the official one.  I bought it here: https://www.amazon.com/dp/B0CFY7SJRN?psc=1&ref=ppx_yo2ov_dt_b_product_details

lemur117, 2023-10-28:
I ran an upgrade on the Orange Pi 5 Plus a couple days ago.  Should I do another one if you just pushed an update?

Joshua-Riek, 2023-10-28:
Yeah that is the official one, you can get the updates on your current system with `sudo apt update && sudo apt upgrade -y`. After the update give your system a moment and reboot. 

If WiFi is still a problem try to reload the kernel module. But i have done a lot of testing and it should be working fine now.
```
sudo modprobe -r 8852be
sudo modprobe -i 8852be
```

Joshua-Riek, 2023-10-28:
> I ran an upgrade on the Orange Pi 5 Plus a couple days ago. Should I do another one if you just pushed an update?

Yes do another update.

lemur117, 2023-10-28:
Thanks, I'll give it a shot.

lemur117, 2023-10-28:
It's working!  After the update and a reboot, I had to do the:

sudo modprobe -r 8852be
sudo modprobe -i 8852be

then it worked!

Thanks for the help!

Joshua-Riek, 2023-10-28:
You may have to run those commands every boot, in the next release i will create a service to do this.

You should be able to put those two lines in /etc/rc.local to make sure the commands are run every boot.

closed 2023-10-28 by Joshua-Riek