Joshua-Riek/ubuntu-rockchip #440: Youtube videos are green screen

Ukhupacha, 2023-11-01:
I'm using an Orange Pi 5, I update the version to 1.29.
Whenever I try to play a video from Youtube, the sound works but the video turns to a green screen
[image]

> chromium-browser 
[4638:4638:1101/130507.812021:ERROR:chrome_browser_cloud_management_controller.cc(162)] Cloud management controller initialization aborted as CBCM is not enabled.
[4681:4681:1101/130507.885169:ERROR:gl_display.cc(520)] EGL Driver message (Error) eglQueryDeviceAttribEXT: eglQueryDeviceAttribEXT
[4681:4681:1101/130507.885285:ERROR:gl_angle_util_vulkan.cc(189)] Failed to retrieve vkGetInstanceProcAddr
[4681:4681:1101/130507.885339:ERROR:vulkan_instance.cc(91)] Failed to get vkGetInstanceProcAddr pointer from ANGLE.
Failed to query video capabilities: Función ioctl no apropiada para el dispositivo

This is the log I have from running chromium on terminal

Penguinator3000, 2023-11-02:
Same happening to me. It is a great OS but has some playback issues... @Joshua-Riek what can we do?

Joshua-Riek, 2023-11-02:
I forgot the details but I'm pretty sure this can't be fixed. It does not support specific video playback resolutions.

Penguinator3000, 2023-11-05:
Yeah. Sometimes 720p video does not work, for me, 1080p works just fine.

Joshua-Riek, 2023-11-05:
Closing as duplicate of https://github.com/Joshua-Riek/ubuntu-rockchip/issues/270

closed 2023-11-05 by Joshua-Riek