Joshua-Riek/ubuntu-rockchip #443: Upgrade to 23.10

Penguinator3000, 2023-11-02:
Hey @Joshua-Riek.
I am truly satisfied with the work that you have done with this port. It helps me develop and do my daily job as if I was in a common PC.

However, I have noticed that some of the issues we all guys have here has to do with third party programs, specifically with dependencies.

I have been investigating and came to the conclusion that 22.04 lacks a lot of aarch64 packages,  probably to make the system more stable (it's a LTS).

I would be truly happy if you could upgrade the port to 23.10, or at least 23.04.

I know it's you doing the work and not me, and I don't know the effort that it requires to do that sort of upgrade.

Thanks a lot.

akhepcat, 2023-11-02:
I'll second this - along with adding the 6.6 kernel - would be super useful for my Opi5+  

As it stands, i can't actually boot the current version due to the kernel crashing on boot, so i'd love to see a 6.6 spin with all the recent packages.

Joshua-Riek, 2023-11-02:
> I'll second this - along with adding the 6.6 kernel - would be super useful for my Opi5+
> 
> As it stands, i can't actually boot the current version due to the kernel crashing on boot, so i'd love to see a 6.6 spin with all the recent packages.

I have 6.6.0 builds working perfectly. What do you mean? Any logs? Please don't just say something does not work without any information that is helpful.

Joshua-Riek, 2023-11-02:
It would be easy to upgrade to Ubuntu 23.10, but you would have no GPU working at all. GPU-related packages would need to be ported, and it's a load of hacks. So, it does not make sense to consider 23.10 or 23.04 when 24.04 is coming out in a few months as the next LTS release. 

However, I may port Ubuntu 23.10 over with a mainline Linux kernel directly from Ubuntu, but not all hardware will work, including the GPU, HDMI, and NPU, and many other features will not work. This is bleeding-edge hardware with poor software support, primarily being worked on by open-source devs in their free time. 

Additionally, this would consume more computational and storage resources, which are limited. Right now, a release is 24GB with 50 artifacts. This will easily double to 48GB and 100 artifacts if I bump every board up to 23.10. On top of this, I need a better way to distribute so many downloads, through a website or some other means.

nyanmisaka, 2023-11-02:
> It would be easy to upgrade to Ubuntu 23.10, but you would have no GPU working at all. GPU-related packages would need to be ported, and it's a load of hacks.

Panfork changes are difficult to maintain and rebase because it contains too many hacks that are hard to understand. This will become a blocker when 24.04 LTS arrives.

We can consider using [mali blob (wayland)](https://github.com/JeffyCN/mirrors/tree/libmali) as an alternative to panfork. I don't know how Friendlyelec did it, but they made libmali work very smoothly on [Ubuntu 22.04 + Gnome desktop](https://wiki.friendlyelec.com/wiki/index.php/NanoPC-T6#Work_with_Ubuntu_22.04_Desktop). This should be the fastest way to get a solid GPU driver and also can avoid green screen issues with youtube videos.

akhepcat, 2023-11-02:
> I have 6.6.0 builds working perfectly. What do you mean? Any logs? Please don't just say something does not work without any information that is helpful.

it's not the 6.6.0 kernel that's crashing.  it's the current 1.2.9 release (server opi5+)  that won't boot for me.  But diagnosing that wasn't the important part, it was the "maybe 6.6.0 will work for me" reason for throwing my upvote.

closed 2023-11-03 by Joshua-Riek