Joshua-Riek/ubuntu-rockchip #444: Installing the adau7002-simple device tree overlay

dblencowe, 2023-11-03:
Hey there!

Firstly, this area of Linux is pretty new to me, so please forgive my noobiness!

I've set up the latest release of your repository on my Rock5a, and am looking to get the Pimoroni Pirateaudio Dual Mic hat working, which is largely built to work with the Raspberry Pi.

As part of the setup process, [detailed here](https://github.com/pimoroni/pirate-audio/blob/master/clip-recorder/README.md), I need to enable the DT overlay `adau7002-simple` ([Link to overlay in rpi repo](https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/overlays/adau7002-simple-overlay.dts))

I tried installing the `linux-generic-modules` package as Google suggests it supplies the driver, but this stopped the device from booting. Presumably, this is because it installed a generic kernel alongside when adding the `linux-firmware` metapackage.

Is it possible for me to get this working on these boards? If so I'd appreciate some pointers on how.

Thanks either way!

Joshua-Riek, 2023-11-03:
You won't be able to install linux-generic-modules, as this does not use an official kernel from ubuntu. As for linux-fimrware i must re-work all the current firmware and split it into new packaging. Then you will be able to download linux-fimware. But this will take time. So many issues are currently open.

Joshua-Riek, 2023-11-03:
You can yoink the dt overlay out of the package, I'm not sure why it would need to be packaged and i assume there are conflicts?

dblencowe, 2023-11-04:
I did wonder if I'd be able to just add the one from the rpi repo to my install but hadn't given it a try as I didn't know what I was doing :joy: 

I'll give it a try in the morning, thanks

closed 2023-11-12 by Joshua-Riek