Joshua-Riek/ubuntu-rockchip #454: the lastest image (desktop version) of rock_5b not support the bcm43456 (bluetooth & wifi).

BaconSoong, 2023-11-08:
use lspci can not find the bcm43456 that attached to the m.2 E-key on the front side of the rock 5b.

Joshua-Riek, 2023-11-08:
Do you have dmesg logs? Was this working previously??

BaconSoong, 2023-11-08:
> Do you have dmesg logs? Was this working previously??

[dmesg_not_support_bcm43456.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13297745/dmesg_not_support_bcm43456.txt)
here u are

BaconSoong, 2023-11-08:
> Do you have dmesg logs? Was this working previously??

just official debian supports it.amrbian and yours does not support.

Joshua-Riek, 2023-11-08:
I don't see any errors in dmesg, but I do know the interface works.

However, I'm working on a major Bluetooth and WiFi firmware update. This may fix the issue. But if it does not, I dont think there is much I can do.

BaconSoong, 2023-11-18:
The problem is still. please check the dmesg.txt.[
[dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13400776/dmesg.txt)
](url)
i find [    8.770177] rk-pcie fe170000.pcie: PCIe Link Fail
[    8.770370] rk-pcie fe170000.pcie: failed to initialize host,as picture shown.
[image]
and lspci does not show the bluetooth&wifi device.
[image]
but, the offical os works ok, Could it be caused by the device tree?

Joshua-Riek, 2023-11-19:
Make sure your PCIe card is fully plugged in, I know it works.

BaconSoong, 2023-11-19:
i'm sure my PCIe card is fully plugged in, because i change the official ubuntu firmware, the bluetooth&wifi works and lspci does not show it, but it exactly works.

Joshua-Riek, 2023-11-19:
I don't know what to say, it works perfectly on both the Rock 5B boards I have.

BaconSoong, 2023-11-19:
> I don't know what to say, it works perfectly on both the Rock 5B boards I have.

all right, May I ask which one you use? I will buy one, thank you very much.

closed 2025-02-17 by github-actions[bot]