Joshua-Riek/ubuntu-rockchip #455: failed to start v2ray-core: not support "system proxy" mode of transparent proxy: does not support to configure system proxy on your OS

BaconSoong, 2023-11-08:
as shown in pdf, "failed to start v2ray-core: not support "system proxy" mode of transparent proxy:
does not support to configure system proxy on your OS."
[Uploading v2rayA.pdf…]()

the v2ray-core and v2ray i have verified on other os are not problem, but here can not.
by the way, this os is the most smooth os!

Joshua-Riek, 2023-11-08:
I don't know what this is and the link I'd broken.

BaconSoong, 2023-11-10:
> I don't know what this is and the link is broken.
v2raya is an open-source project on GitHub. It serves as the graphical user interface (GUI) for v2ray-core, and it appears that v2raya can not now control v2ray-core.

Joshua-Riek, 2023-11-10:
I see. Please provide reproducible steps of the issue.

124ffee, 2023-11-11:
Without using 'v2ray-core,' as I've had trouble with it. Instead, I've been able to use 'Clash for Windows,' but with the previous open-source code.

124ffee, 2023-11-11:
I've found a complete tutorial in this video, including creating desktop shortcuts: https://www.youtube.com/watch?v=VOlWdNZAq_o&list=PLHGQWf4FUitSG5srcvJqdl-FMJYMG6wZc&index=8&t=660s

Joshua-Riek, 2023-11-13:
Closing as not enough info to reproduce the problem.

closed 2023-11-13 by Joshua-Riek