Joshua-Riek/ubuntu-rockchip #456: No name RK3588 Board

irlkitcom, 2023-11-09:
Hello,

I have a no name RK3588 board. It came with Android 12 installed. I contacted the seller and they don't have an image for it for Linux. Is there any chance this board could run Ubuntu?

Joshua-Riek, 2023-11-09:
Sure if you reverse engineer the device tree. But I will not help with this.

closed 2023-11-09 by Joshua-Riek

ferrellsl, 2023-11-09:
If you provided a good photo of your board we might be able to tell what it was cloned from.  More than likely it's a copy of a 5 or 5b.

irlkitcom, 2023-11-10:
[image]

ferrellsl, 2023-11-10:
That looks a lot like a Mekotronics R58 Mini.  I own a Mini and it came pre-loaded with Android TV OS 12.  Do you get any hints as to what hardware or OS you're running when you look under the Settings->About?
Here's what mine says:

irlkitcom, 2023-11-10:
Everything in the About section just says RK3588_S, even though the chip on the board is 100% an RK3588 non S variant. It's running Android 12 Tablet version.