Joshua-Riek/ubuntu-rockchip #460: clone E-MMC to MICROSD

rusudanion, 2023-11-11:
Hello i want to make an BACKUP of actual boot image . I have an opi5 plus with 256emmc and i want  backup entire flash to  microsd, or make a backup image of entire flash. how can i make ? thankyou

rusudanion, 2023-11-13:
thankyou but i need to know the command line.i can,'t find.

spele1100, 2023-11-13:
> thankyou but i need to know the command line.i can,'t find.

If you are asking the dd command, it's something like:
sudo dd if=/dev/mmcblk0 of=backup.img bs=1M count=262144 progress=status
For the count you should better calculate by yourself, otherwise just take out the bs and count, wait it until finish.
Details can reference to some online tutorial: https://linuxopsys.com/topics/linux-dd-command-with-examples
Just make sure you don't mess up the "if" and "of"

closed 2023-11-13 by Joshua-Riek