Joshua-Riek/ubuntu-rockchip #461: Clone emmc of orange pi5b board another result MAC same

spele1100, 2023-11-13:
I have multiple orange pi 5 board, one of the board installed all the software I needed. I had clone that emmc of the board as a source image a put that to the others, but results the MAC address of eth0 and wlan0 are all the same as source image.

spele1100, 2023-11-13:
Could you share the location on where the the MAC address is stored or which script had run while fresh install so that I can work it around?

rusudanion, 2023-11-13:
> I have multiple orange pi 5 board, one of the board installed all the software I needed. I had clone that emmc of the board as a source image a put that to the others, but results the MAC address of eth0 and wlan0 are all the same as source image.

hello can you give me the command line for clone E-MMC  . thankyou

spele1100, 2023-11-13:
> hello can you give me the command line for clone E-MMC . thankyou

Replied in your thread.

Joshua-Riek, 2023-11-13:
> Could you share the location on where the the MAC address is stored or which script had run while fresh install so that I can work it around?

I don't know where this is located, will need some research.

samirsommaior, 2023-11-13:
hello I also would like if you can share the command line for clone E-MMC, thanks

spele1100, 2023-11-13:
> hello I also would like if you can share the command line for clone E-MMC, thanks

Please refer to: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/460#issuecomment-1808316485

theflyingape, 2023-11-14:
Does [this article](https://forum.armbian.com/topic/28005-mac-adress-change-everyboot-orange-5) work for you?

spele1100, 2023-11-14:
> Does [this article](https://forum.armbian.com/topic/28005-mac-adress-change-everyboot-orange-5) work for you?

I was trying to modify the /etc/network/interface, but unfortunately the file is no longer exist at jammy. Also tried the macchanger which is not working too.

Currently I made some a workaround script to change the following stuff,
/

closed 2023-11-14 by spele1100