Joshua-Riek/ubuntu-rockchip #463: Enabling SPI interface for the Rock5a

dblencowe, 2023-11-13:
Hi there,

I'm trying to load the spi interface on the Rock 5A so that I can use the python-spidev library, unfortunately I've not had any success enabling the overlays in the /boot/dtbs/overlays/ directory.

Is this possible with this kernel & board?

dblencowe, 2023-11-13:
I'm familiar with the process from my ongoing trials & tribulations getting this Audio hat to work :joy:

I've been unable to decipher the meaning of the segments in the below filenames, which has left me guessing; Are you able to shed any light on this?

```
rk3588-spi0-m1-cs0-spidev.dtbo
rk3588-spi0-m1-cs1-spidev.dtbo
rk3588-spi0-m2-cs0-cs1-spidev.dtbo
rk3588-spi0-m2-cs0-spidev.dtbo
rk3588-spi0-m2-cs1-spidev.dtbo
rk3588-spi1-m1-cs0-spidev.dtbo
rk3588-spi1-m1-cs1-mcp2515-8mhz.dtbo
rk3588-spi1-m1-cs1-spidev.dtbo
rk3588-spi4-m0-cs1-spidev.dtbo
rk3588-spi4-m1-cs0-cs1-spidev.dtbo
rk3588-spi4-m1-cs0-spidev.dtbo
rk3588-spi4-m1-cs1-spidev.dtbo
rk3588-spi4-m2-cs0-spidev.dtbo
rock-5a-spi-nor-flash.dtbo
```

Including `rk3588-spi0-m1-cs0-spidev` seems to have added a /dev/rkspi-dev2 device, but I still don't see a /dev/spidev*.*

Thanks for your help, It is greatly appreciated

Joshua-Riek, 2023-11-13:
Weird, I've never seen `rkspi-dev2` before, maybe its the same as `spidev`? Do you know for sure which pin / overlay you need for spi?

dblencowe, 2023-11-13:
I tried symlinking the rkspi-dev2 file to the location that spidev expects it, but only got an error: [Errno 25] Inappropriate ioctl for device

I'm unsure which bus / pin I'm looking for, on the PI it would be bus 0, chip 1. I'm trying to get the below code working for an st7789 display:

https://gist.github.com/dblencowe/419ad9d585a1d7d8be723ac2b14da4fc

Joshua-Riek, 2023-11-13:
I checked the overlay source code, and its using `rockchip,spidev`, so I think it should work. But i dont know which overlay you would need. 

Maybe try: `rk3588-spi4-m1-cs0-cs1-spidev`

dblencowe, 2023-11-14:
Hey Joshua,

Thanks for your help, this looks to have worked. Not sure why it didn't when I loaded them all, but I suspect that will be because I caused some sort of conflict.

closed 2023-11-14 by dblencowe