Joshua-Riek/ubuntu-rockchip #466: Linux on ssd

Gazgiza, 2023-11-14:
Is it possible to install the os on an ssd and have it as the main disk?

Joshua-Riek, 2023-11-14:
Of course, how to do this depends on the board being used.

Gazgiza, 2023-11-15:
How can I manage to do this?

Joshua-Riek, 2023-11-15:
You need to provide way more information.

Gazgiza, 2023-11-18:
I have an sd card with 64 gb on it an want my os to be installed on an ssd plugged in the m.2 slot but idk how to make this. There was something in the readme.md but it disapered.

FreeWilly2022, 2023-11-23:
I was also looking for the directions on how to copy to an SSD, and I found them here:
https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/435

paulwratt, 2023-11-23:
when you "boot" the linux kernel you have to pass an alternate / new _root device_ as the mount point. (`cmdline.txt` on a RPi, or **_boot options_** in a bootloader menu)

Alternately, once you have booted (from sd-card) you can `cd` into the root of the new device you want to use as the root filesystems, `mount --bind /dev (here/)dev` (for `/proc` & `/sys` too) the `chroot` - if you know where the SSD is going to be in the `/dev`ices then you can add _those commands_ to the end of your sd-card `init` boot procedure to automate the change (NOTE: both drives need to be using same / similar kernel)

develperbayman, 2023-12-24:
honestly the easiest way i did it was use a ssd usb adapter and then just flashed the image also if it dont boot i used armbians orangepi-config to update the spi flash to boot then i flashed the image .....i know thats a round about way but it was pretty easy @ joshua-riek you should consider adding the orangepi-config script (i dont think its on this distro? never had to check lol)

closed 2023-12-25 by Joshua-Riek