Joshua-Riek/ubuntu-rockchip #468: Orange Pi 5 unable to hotplug Headset Mic Jack in Pulseaudio

arkadijs, 2023-11-16:
There seems to be a problem with 5.10.160-rockchip es8388 kernel driver reporting headset mic as always unplugged on Orange Pi 5 unless the mic is plugged before the board is booted. Unplugging a working mic and plugging it back is reported correctly.

The mic itself do work regardless with raw Alsa but Pulseaudio is confused by port `available: no` status -- silent recording.

```
$ amixer -c2 cget numid=37,iface=CARD,name='Headset Mic Jack'
numid=37,iface=CARD,name='Headset Mic Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off

$ pacmd list-sources
analog-input-headset-mic: Headset Microphone (priority 8800, latency offset 0 usec, available: no)
```

Joshua-Riek, 2023-11-17:
Interesting, thanks for reporting this. I see two things that could be the issue, either the headphone det gpio is not correct in the device tree, or a pluseaudio config is required.

arkadijs, 2023-11-17:
I'd say it's functional but flaky. I left the board overnight with (headset) mic attached and it didn't work in the morning -- still reported as connected though, but it did work yesterday. Unplugged, plugged back -- it works again.

To clear my initial statement: reporting mic port status works if headset is attached before board is booted.

Thank you for looking into this.

develperbayman, 2023-12-29:
please consider fixing the built in microphone im using my opi5 for artificial intelligence and its weird trying to use a external mic for a application that is supposed to be natural as the board is hidden i would greatly appreciate it

closed 2025-02-17 by github-actions[bot]