Joshua-Riek/ubuntu-rockchip #472: Cannot get access to UART or any other GPIO function

emusman-lab, 2023-11-18:
Hello!
I am using orangepi 5B plus with latest ubuntu 22.01 from here (https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.30 ).
I have attached arduino to serial 0 and want to read the data. However i cannot find /dev/ttyS0 in terminal here is my output of **dmesg | grep tty**.
`[    2.147534] Kernel command line: root=UUID=7353bfc8-aee0-4b74-a1f6-d87826c03972 rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 systemd.unified_cgroup_hierarchy=0 quiet splash plymouth.ignore-serial-consoles
[    2.551969] printk: console [tty1] enabled
[    2.587866] Registered FIQ tty driver
[    3.579405] printk: console [ttyFIQ0] enabled
[    3.579504] Registered fiq debugger ttyFIQ0
[    4.509589] febc0000.serial: ttyS9 at MMIO 0xfebc0000 (irq = 113, base_baud = 1500000) is a 16550A
[    7.838861] systemd[1]: Created slice Slice /system/serial-getty.
[    8.917919] [dhd] [INIT] logset:8 is preserve/chatty
[    8.919261] [dhd] [INIT] logset:10 is preserve/chatty
[   10.270645] [dhd] [INIT] logset:8 is preserve/chatty
[   10.271863] [dhd] [INIT] logset:10 is preserve/chatty
[   21.247453] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2`
I have followed your instructions from these two links 
(https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/212)
to modify the ubuntuEnv.txt file but still i get the same output from the **dmesg | grep tty** as above.
here are the contents of my ubuntuEnv.txt file
`bootargs=root=UUID=7353bfc8-aee0-4b74-a1f6-d87826c03972 rootfstype=ext4 rootwai>
fdtfile=rk3588s-orangepi-5b.dtb
overlay_prefix=orangepi-5
overlays=rk3588-uart0-m2.dtb`
Here is the output of **ls /boot/firmware/dtbs/overlays/ | grep rk3588-uart0**
`rk3588-uart0-m2.dtbo`

Please let me know what i am doing wrong here

Joshua-Riek, 2023-11-18:
Looks like a typo, don't add the `.dtbo` in ubuntuEnv.txt

emusman-lab, 2023-11-18:
Right Thanks for your response Joshua, and thank for your contribution for the community.
I'll try this.

closed 2025-02-17 by github-actions[bot]