Joshua-Riek/ubuntu-rockchip #473: The speed of upgrade is so slow, how to improve it?

BaconSoong, 2023-11-18:
as picture shown, the speed is 8146B/s.
[image]

BaconSoong, 2023-11-18:
and how to check i have upgraded successfully?

Joshua-Riek, 2023-11-18:
There should be ubuntu mirrors available for different regions, I'm unable to look up an exact list right now. But it should be easily doable.

Joshua-Riek, 2023-11-19:
You can change the `/etc/sources.list` file to use a ubuntu mirror for your region.

For example, users in China may want to use `repo.huaweicloud.com` for better download speeds.
```
deb http://repo.huaweicloud.com/ubuntu-ports/ jammy main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy main restricted universe multiverse

deb http://repo.huaweicloud.com/ubuntu-ports/ jammy-security main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy-security main restricted universe multiverse

deb http://repo.huaweicloud.com/ubuntu-ports/ jammy-updates main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy-updates main restricted universe multiverse

deb http://repo.huaweicloud.com/ubuntu-ports/ jammy-backports main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy-backports main restricted universe multiverse
```

More information: https://askubuntu.com/questions/39922/how-do-you-select-the-fastest-mirror-from-the-command-line

BaconSoong, 2023-11-19:
thank u very much!

closed 2023-11-20 by Joshua-Riek