Joshua-Riek/ubuntu-rockchip #474: Booting from USB with existing NVME-install (opi5+)

bagong, 2023-11-18:
Hi,

is it possible too boot from USB if there already is a working install on a NVME SSD (as is possible from SD)? I think I read somewhere it should be possible, but it doesn't seem to be working for me? It's a Samsung fitt USB-drive, I like using it for a portable OS...

Thanks for your consideration
.r.

Joshua-Riek, 2023-11-18:
It looks like the USB 3.0 ports do not work in U-Boot. I'm looking at the Rockchip documentation and will try to come up with a fix.

bagong, 2023-11-18:
Thank you, you are very helpful!

Joshua-Riek, 2023-11-18:
I was able to get USB 3.0 working with mainline U-Boot but not the BSP version. I think USB 3.0 may be broken with the BSP bootloader. This will require more research.

bagong, 2023-11-18:
Okay, thank you for confirming, and let me know if I can help!

Joshua-Riek, 2023-11-18:
Since USB 3.0 works with mainline U-Boot, I will only spend a little time on this. Hopefully, in 1-2 months, we can update boards to use U-Boot v2024.01, but we will need to ensure it's stable and not cause any problems.

bagong, 2023-11-18:
I understand, I will use an SD-card for the time being. Good to know this behavior is reproducible and I am not doing anything wrong. Thank you!