Joshua-Riek/ubuntu-rockchip #480: Netflix not working on Chromium

emfol, 2023-11-22:
I have successfully installed Ubuntu-Rockchip on my Orange Pi 5B but I can't stream any titles on Netflix. I can login, I can browse titles (even title previews work) but when I try to effectively play a title, something goes wrong and I'm presented with an `M7111-1331` error. The issue only happens on Netflix. I have already tested other streaming services like DisneyPlus and they all work great.

Any idea on what might be causing the issue?

Another thing I noticed is that all videos in my Twitter timeline go completely green... They start playing but after a few seconds they go green... I can still hear the video playing but only a green canvas is presented.

There is always a chance that I might have acquired a faulty hardware but I thought it would be worth sharing this here in case anyone experiences the same issue.

Thanks in advance.

Joshua-Riek, 2023-11-22:
Green videos are because of unsupported resolutions, this will not be fixed

edp1096, 2023-11-24:
Hello @emfol.
The issue about `M7111-1331` on netflix, @Joshua-Riek answered several time.
I solved it by the answer at #282

emfol, 2023-11-24:
Thanks @Joshua-Riek and @edp1096.
Changing Chromium's user-agent string got Netflix working great.
It's amazing what this RK3588-tailored version of Ubuntu has accomplished for my Orange Pi 5... I am impressed.
Thanks a lot!

closed 2023-11-24 by emfol