Joshua-Riek/ubuntu-rockchip #483: How to enable eBPF on OPi5 and V1.30 images

Zhu-HH, 2023-11-27:
Thank you for your contribution to the community!

I want to learn eBPF on the OPi5 board by manually compiling the kernel, but after installing and restarting the kernel, I found that I cannot access the system interface, but SSH can connect.

r0ysue, 2023-12-03:
eBPF Kernel requirements：

https://github.com/iovisor/bcc/blob/master/INSTALL.md#kernel-configuration

Zhu-HH, 2023-12-03:
here are the exact changes ：

scripts/config --enable CONFIG_BPF_LSM
scripts/config --enable CONFIG_KGDB
scripts/config --enable CONFIG_FUNCTION_PROFILER
scripts/config --enable CONFIG_FTRACE_SYSCALLS
scripts/config --enable CONFIG_BPF_KPROBE_OVERRIDE
scripts/config --enable CONFIG_TRACE_EVENT_INJECT
scripts/config --enable CONFIG_HIST_TRIGGERS
scripts/config --enable CONFIG_SCHED_TRACER
scripts/config --enable CONFIG_IRQSOFF_TRACER
scripts/config --enable CONFIG_FUNCTION_TRACER
scripts/config --enable CONFIG_STACK_TRACER
scripts/config --disable CONFIG_MAGIC_SYSRQ
scripts/config --enable CONFIG_IKHEADERS
scripts/config --enable CONFIG_DEBUG_INFO_BTF
scripts/config --enable CONFIG_DEBUG_INFO
scripts/config --disable DEBUG_INFO_REDUCED
scripts/config --disable DEBUG_INFO_COMPRESSED
scripts/config --disable DEBUG_INFO_SPLIT
scripts/config --disable DEBUG_INFO_DWARF4
scripts/config --disable GDB_SCRIPTS
scripts/config --disable DEBUG_EFI
scripts/config --disable ARM64_RELOC_TEST

scripts/config --enable CONFIG_BPF_SYSCALL
scripts/config --enable CONFIG_BPF_JIT
scripts/config --enable CONFIG_HAVE_BPF_JIT
scripts/config --enable CONFIG_HAVE_EBPF_JIT
scripts/config --enable CONFIG_MODULES
scripts/config --enable CONFIG_BPF
scripts/config --enable CONFIG_BPF_EVENTS
scripts/config --enable CONFIG_PERF_EVENTS
scripts/config --enable CONFIG_HAVE_PERF_EVENTS
scripts/config --enable CONFIG_PROFILING
scripts/config --enable CONFIG_BPF_UNPRIV_DEFAULT_OFF
scripts/config --enable CONFIG_CGROUP_BPF
scripts/config --enable CONFIG_BPFILTER
scripts/config --module CONFIG_BPFILTER_UMH
scripts/config --module CONFIG_NET_CLS_BPF
scripts/config --module CONFIG_NET_ACT_BPF
scripts/config --enable CONFIG_BPF_STREAM_PARSER
scripts/config --enable CONFIG_LWTUNNEL_BPF
scripts/config --module CONFIG_NETFILTER_XT_MATCH_BPF
scripts/config --enable CONFIG_IPV6_SEG6_BPF
scripts/config --enable CONFIG_KPROBE_EVENTS
scripts/config --enable CONFIG_KPROBES
scripts/config --enable CONFIG_HAVE_KPROBES
scripts/config --enable CONFIG_HAVE_REGS_AND_STACK_ACCESS_API
scripts/config --enable CONFIG_KPROBES_ON_FTRACE
scripts/config --enable CONFIG_FPROBE
scripts/config --enable CONFIG_BPF_KPROBE_OVERRIDE
scripts/config --enable CONFIG_UPROBE_EVENTS
scripts/config --enable CONFIG_ARCH_SUPPORTS_UPROBES
scripts/config --enable CONFIG_UPROBES
scripts/config --enable CONFIG_MMU
scripts/config --enable CONFIG_TRACEPOINTS
scripts/config --enable CONFIG_HAVE_SYSCALL_TRACEPOINTS
scripts/config --enable CONFIG_BPF_LSM
scripts/config --enable CONFIG_BPF_LIRC_MODE2
scripts/config --module CONFIG_VSOCKETS
scripts/config --module CONFIG_VSOCKETS_DIAG
scripts/config --enable CONFIG_VHOST_VSOCK
scripts/config --enable CONFIG_VHOST_NET
scripts/config --enable CONFIG_VSOCKMON
scripts/config --disable SERIAL_CONEXANT_DIGICOLOR
scripts/config --disable SERIAL_SPRD
scripts/config --disable CONFIG_KPROBES_SANITY_TEST
scripts/config --enable KGDB_HONOUR_BLOCKLIST
scripts/config --enable BPF_JIT_ALWAYS_ON
scripts/config --disable CONFIG_TRACEFS_DISABLE_AUTOMOUNT
scripts/config --enable CONFIG_KALLSYMS_ALL

> I don't know what eBPF is or what modifications made to the kernel in that made the system interface no longer accessible.
> 
> Can you provide the exact changes?

gclawes, 2024-08-05:
@Joshua-Riek it looks like this was missing in the upstream `linux-raspi` as well, fix pending release:
https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/2065829
https://launchpad.net/ubuntu/+source/linux-raspi/6.8.0-1009.10

gclawes, 2024-08-06:
> eBPF Kernel requirements：
> 
> https://github.com/iovisor/bcc/blob/master/INSTALL.md#kernel-configuration

Looks like this is the full list of configs: https://github.com/iovisor/bcc/blob/master/docs/kernel_config.md

In particular, the BTF features seem to be required for "Compile Once - Run Everywhere" eBPF programs like those used by falco:
https://falco.org/docs/event-sources/kernel/#modern-ebpf-probe
https://nakryiko.com/posts/bpf-portability-and-co-re/

Also, it looks like `bpftool` is missing for the particular kernel versions distributed with these images:
```
ubuntu@rk01:~$ uname -a
Linux rk01 6.1.0-1021-rockchip #21-Ubuntu SMP Mon Jul 29 03:52:32 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@rk01:~$ sudo bpftool feature probe kernel | grep -q "map_type ringbuf is available" && echo "true" || echo "false"
WARNING: bpftool not found for kernel 6.1.0-1021

  You may need to install the following packages for this specific kernel:
    linux-tools-6.1.0-1021-rockchip
    linux-cloud-tools-6.1.0-1021-rockchip

  You may also want to install one of the following packages to keep up to date:
    linux-tools-rockchip
    linux-cloud-tools-rockchip
false
```

Joshua-Riek, 2024-08-06:
I enabled the missing kernel modules, though I have not tested to confirm if this works yet.

See: https://github.com/Joshua-Riek/linux-rockchip/commit/73c50d1756568e9c2b1da19b875ea112fa998c00

Joshua-Riek, 2024-08-13:
Ah of course, I forgot to enable bpftool in the kernel packaging.

https://github.com/Joshua-Riek/linux-rockchip/commit/fb5326b94b5a52f522ba6c81d2ae80eb5c6cf86b

Joshua-Riek, 2024-08-13:
I enabled support for BTF in the kernel for Ubuntu 22.04 as well. I see that bpftool is now in the built package, will do some testing later tonight and see if it works.

Joshua-Riek, 2024-08-15:
I pushed a kernel update for Jammy and Noble systems. Please update and install the `linux-rockchip-tools` package.

closed 2024-08-15 by Joshua-Riek