Joshua-Riek/ubuntu-rockchip #485: Building v4l2loopback-dkms, dependence on kernel 5.15 and gcc12

jardinimf, 2023-11-27:
1st I liked this distribution, the experience out-of-box is on par with the one of a _full-pc_, really good. (I'll open a ticket asking for the minimal-Desktop version later)

I'm using the Orange Pi 5 Plus for a client project. They need the HDMI Rx feature, it worked oob, and they also need that feature exposed for use in the Browser, and for that I'm using the package v4l2loopback.

I've already tested it on the Orange's Ubuntu distribution and it's working. Trying the same on your distro I caught that the package dkms, that is a dependence of v4l2loopback-dkms, is trying to install gcc-12 and kernel-headers for 5.15 instead of using the gcc-11 and linux-header 5.10 from rockchip that are already installed.

P.S.: I've used the oficial RKDevTool to install your distro, worked like a charm and is faster than using the SD, the app have an english translation, it only needs changing the _Selected=1_ to _Selected=1_ in the key _[Language]_ in the file **config.ini**

Joshua-Riek, 2023-11-27:
Can you provide the exact commands and errors you received on the distro from orangepi and my own?

The kernel situation is weird right now, as the only fully working kernel that supports the hardware entirely is an Android hack.

jardinimf, 2023-11-27:
Hello Joshua.

- _The kernel situation is weird right now, as the only fully working kernel that supports the hardware entirely is an Android hack._
Understand, I've worked on a previous project with another CPU with the same issue, they usually only provide hacked Android samples.

- _Can you provide the exact commands and errors you received on the distro from orangepi and my own?_
Sure, I can provide the error, np.

**_Commad_**
`$ sudo apt install v4l2loopback-dkms v4l2loopback-utils`

**_Output_**
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  cpp-12 dctrl-tools dkms gcc-12 libasan8 libgcc-12-dev libtsan2 linux-headers-5.15.0-88 linux-headers-5.15.0-88-generic
  linux-headers-generic
Suggested packages:
  gcc-12-locales cpp-12-doc debtags menu gcc-12-doc
The following NEW packages will be installed:
  cpp-12 dctrl-tools dkms gcc-12 libasan8 libgcc-12-dev libtsan2 linux-headers-5.15.0-88 linux-headers-5.15.0-88-generic
  linux-headers-generic v4l2loopback-dkms v4l2loopback-utils
0 upgraded, 12 newly installed, 0 to remove and 0 not upgraded.
Need to get 2.340 B/50,1 MB of archives.
After this operation, 216 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Ign:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-generic arm64 5.15.0.88.85
Err:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-generic arm64 5.15.0.88.85
  404  Not Found [IP: 2620:2d:4000:1::19 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/l/linux-meta/linux-headers-generic_5.15.0.88.85_arm64.deb  404  Not Found [IP: 2620:2d:4000:1::19 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

For Orange Pi Ubuntu
**_Commad_**
`$ sudo dpkg -i \opt\linux-headers-legacy-rockchip-rk3588_1.0.4_arm64.deb`
`$ sudo apt install v4l2loopback-dkms v4l2loopback-utils`

Joshua-Riek, 2023-11-27:
That is strange, the command on my system fails because any can't find the kernel headers. 

I don't know why this is not the case on the Orange Pi system. But what is the output on the Orange Pi ubuntu image?

jardinimf, 2023-11-27:
It compiles the v4l2loopback-dkms in a few seconds and the module works just fine.

Their apt source.list points to Huawei
```
deb http://repo.huaweicloud.com/ubuntu-ports/ jammy main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy main restricted universe multiverse

deb http://repo.huaweicloud.com/ubuntu-ports/ jammy-security main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy-security main restricted universe multiverse

deb http://repo.huaweicloud.com/ubuntu-ports/ jammy-updates main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy-updates main restricted universe multiverse

deb http://repo.huaweicloud.com/ubuntu-ports/ jammy-backports main restricted universe multiverse
#deb-src http://repo.huaweicloud.com/ubuntu-ports/ jammy-backports main restricted universe multiverse
```

I'll try use their .deb (http://repo.huaweicloud.com/ubuntu-ports/pool/universe/v/v4l2loopback/)

Joshua-Riek, 2023-11-27:
Maybe try to install the dkms package first, then use apt to install v4l2loopback-dkms?

jardinimf, 2023-11-27:
- _Maybe try to install the dkms package first, then use apt to install v4l2loopback-dkms?_
It is the exact same error.

```
$ sudo apt install dkms

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  cpp-12 dctrl-tools gcc-12 libasan8 libgcc-12-dev libtsan2
  linux-headers-5.15.0-88 linux-headers-5.15.0-88-generic
  linux-headers-generic
Suggested packages:
  gcc-12-locales cpp-12-doc debtags menu gcc-12-doc
The following NEW packages will be installed:
  cpp-12 dctrl-tools dkms gcc-12 libasan8 libgcc-12-dev libtsan2
  linux-headers-5.15.0-88 linux-headers-5.15.0-88-generic
  linux-headers-generic
0 upgraded, 10 newly installed, 0 to remove and 0 not upgraded.
Need to get 2.340 B/50,1 MB of archives.
After this operation, 216 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Ign:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-generic arm64 5.15.0.88.85
Err:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-generic arm64 5.15.0.88.85
  404  Not Found [IP: 2620:2d:4000:1::16 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/l/linux-meta/linux-headers-generic_5.15.0.88.85_arm64.deb  404  Not Found [IP: 2620:2d:4000:1::16 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

jardinimf, 2023-11-27:
Your distro have the correct headers (5.10) installed, but the packages on your source.list are pointing to 5.15

```
$ apt list --installed | grep linux

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

binutils-aarch64-linux-gnu/jammy-updates,jammy-security,now 2.38-4ubuntu2.3 arm64 [installed,automatic]
console-setup-linux/jammy,now 1.205ubuntu3 all [installed]
libselinux1/jammy,now 3.3-1build2 arm64 [installed]
linux-base/jammy,now 4.5ubuntu9 all [installed,automatic]
linux-firmware/jammy-updates,now 20220329.git681281e4-0ubuntu3.22 all [installed]
linux-headers-5.10.160-rockchip/jammy,now 5.10.160-21 arm64 [installed,automatic]
linux-headers-rockchip-5.10/jammy,now 5.10.160-21 arm64 [installed,automatic]
linux-image-5.10.160-rockchip/jammy,now 5.10.160-21 arm64 [installed,automatic]
linux-image-rockchip-5.10/jammy,now 5.10.160-21 arm64 [installed,automatic]
linux-libc-dev/jammy-updates,jammy-security,now 5.15.0-88.98 arm64 [installed,automatic]
linux-rockchip-5.10/jammy,now 5.10.160-21 arm64 [installed]
linux-sound-base/jammy,now 1.0.25+dfsg-0ubuntu7 all [installed,automatic]
pptp-linux/jammy,now 1.10.0-1build3 arm64 [installed,automatic]
util-linux/jammy,now 2.37.2-4ubuntu3 arm64 [installed]
```

Joshua-Riek, 2023-11-27:
Oh libc dev is patched for winedevine use. You can try and purge that package then maybe dkms will work as expected.

jardinimf, 2023-11-27:
Removed libc6-dev (`apt autoremove --purge libc6-dev`) then the dkms and v4l2loopback installs with success.

The module works, it list on browser as "loopcam"
```
$ sudo modprobe v4l2loopback video_nr=20 card_label=loopcam exclusive_caps=1
$ gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video20
```

But fails with `gst-launch-1.0 v4l2src source=/dev/video0 ! ... ! v4l2sink device=/dev/video20`, but that is for another time.

Thank you Joshua

closed 2025-02-17 by github-actions[bot]