Joshua-Riek/ubuntu-rockchip #490: Display artifacts over Type C when shutting down. Orange PI 5 Plus.

TuTuc0, 2023-11-29:
When a monitor is connected over Type C and OPI5+ is shutting down, the video output looks "broken and disassembled". It shows many small horizontal lines, and colors are shifted like heavy chromatic aberration effect. 
The artifacts show up always when shutting down or restarting OPI5+. Video output during boot looks normal.
I have also tried other refresh rates. (165, 120, 90, 60hz)

Specs:
OPI5+
Monitor: Gigabyte M27Q, 1440p, 165Hz
Only power port and type c port are occupied. (HDMI ports are empty).
Few USB peripherals are connected to monitor over usb (mouse, keyboard, speakers, camera).

Monitor and Type C cable works fine without issues on other devices. 

Attaching photo of display.
[image]

jardinimf, 2023-11-29:
My 2cents.
The USB 3.0 Type-C/DP (near the MaskROM button) is an OTG port, when shutdown the driver may be downgrading back to generic USB port. It's expected or is an issue to be reported to the kernel maintainer upstream.

Joshua-Riek, 2023-12-25:
The above description sounds accurate, but since the kernel is from Rockchip this will not be fixed. It's to much work to properly fix in an already hacked up kernel.

closed 2023-12-25 by Joshua-Riek