Joshua-Riek/ubuntu-rockchip #496: 6.7rc3 mainline i think HDMI actually works now but it is missing something

Shivansps, 2023-12-02:
The HDMI output now is getting initialized and i get a black screen output, before there was no signal. Here is the current output of dmesg on 6.7rc3 mainline

ubuntu@ubuntu:~$ dmesg | grep hdmi
[    0.057776] platform fdd90000.vop: Fixed dependency cycle(s) with /hdmi@fde80000/ports/port@0/endpoint
[    0.069638] platform hdmi-con: Fixed dependency cycle(s) with /hdmi@fde80000/ports/port@1/endpoint
[    2.362252] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy init success
[    2.366306] dwhdmi-rockchip fde80000.hdmi: no ddc property found
[    2.368585] dwhdmi-rockchip fde80000.hdmi: registered ddc I2C bus driver
[    2.369475] rockchip-drm display-subsystem: bound fde80000.hdmi (ops dw_hdmi_rockchip_ops [rockchipdrm])
[    2.399587] dwhdmi-rockchip fde80000.hdmi: read_hpd result: 1
[    2.399601] dwhdmi-rockchip fde80000.hdmi: i2c xfer: num: 2, addr: 0x50
[    2.399628] dwhdmi-rockchip fde80000.hdmi: xfer: num: 1/2, len: 1, flags: 0x0
[    2.399633] dwhdmi-rockchip fde80000.hdmi: i2c write done!
[    2.399637] dwhdmi-rockchip fde80000.hdmi: xfer: num: 2/2, len: 1, flags: 0x1
[    2.400089] dwhdmi-rockchip fde80000.hdmi: i2c xfer: num: 2, addr: 0x50
[    2.400108] dwhdmi-rockchip fde80000.hdmi: xfer: num: 1/2, len: 1, flags: 0x0
[    2.400131] dwhdmi-rockchip fde80000.hdmi: i2c write done!
[    2.400139] dwhdmi-rockchip fde80000.hdmi: xfer: num: 2/2, len: 128, flags: 0x1
[    2.458486] dwhdmi-rockchip fde80000.hdmi: i2c xfer: num: 2, addr: 0x50
[    2.458510] dwhdmi-rockchip fde80000.hdmi: xfer: num: 1/2, len: 1, flags: 0x0
[    2.458521] dwhdmi-rockchip fde80000.hdmi: i2c write done!
[    2.458528] dwhdmi-rockchip fde80000.hdmi: xfer: num: 2/2, len: 128, flags: 0x1
[    2.515197] dwhdmi-rockchip fde80000.hdmi: got edid: width[12] x height[8]
[    2.519501] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[    2.520312] dwhdmi-rockchip fde80000.hdmi: Non-CEA mode used in HDMI
[    2.520321] dwhdmi-rockchip fde80000.hdmi: final pixclk = 33900000
[    2.520329] dwhdmi-rockchip fde80000.hdmi: final tmdsclk = 33900000
[    2.520341] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: bus_width:0x52c38,bit_rate:339000
[    2.520351] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:52c38 rate:339000
[    2.520700] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
[    2.520958] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy lane locked!
[    2.520969] dwhdmi-rockchip fde80000.hdmi: sink has audio support
[    2.520977] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_qp_setup HDMI mode
[    2.521054] dwhdmi-rockchip fde80000.hdmi: don't use dsc mode
[    2.727586] dwhdmi-rockchip fde80000.hdmi: dw hdmi qp use tmds mode
[    2.727601] dw_hdmi_dclk_set vp0
[    2.727618] dwhdmi-rockchip fde80000.hdmi: [drm:dw_hdmi_dclk_set [rockchipdrm]] *ERROR* failed to get dclk_vp0
[    2.740877] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[    2.834285] dwhdmi-rockchip fde80000.hdmi: i2c xfer: num: 2, addr: 0x50
[    2.834299] dwhdmi-rockchip fde80000.hdmi: xfer: num: 1/2, len: 1, flags: 0x0
[    2.834304] dwhdmi-rockchip fde80000.hdmi: i2c write done!
[    2.834307] dwhdmi-rockchip fde80000.hdmi: xfer: num: 2/2, len: 1, flags: 0x1
[    2.834792] dwhdmi-rockchip fde80000.hdmi: i2c xfer: num: 2, addr: 0x50
[    2.834801] dwhdmi-rockchip fde80000.hdmi: xfer: num: 1/2, len: 1, flags: 0x0
[    2.834806] dwhdmi-rockchip fde80000.hdmi: i2c write done!
[    2.834809] dwhdmi-rockchip fde80000.hdmi: xfer: num: 2/2, len: 128, flags: 0x1
[    2.888422] dwhdmi-rockchip fde80000.hdmi: i2c xfer: num: 2, addr: 0x50
[    2.888436] dwhdmi-rockchip fde80000.hdmi: xfer: num: 1/2, len: 1, flags: 0x0
[    2.888443] dwhdmi-rockchip fde80000.hdmi: i2c write done!
[    2.888448] dwhdmi-rockchip fde80000.hdmi: xfer: num: 2/2, len: 128, flags: 0x1
[    2.948413] dwhdmi-rockchip fde80000.hdmi: got edid: width[12] x height[8]
[    2.948808] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[    7.978142] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[    7.997595] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[    8.062639] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[    8.164514] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0
[   28.756004] dwhdmi-rockchip fde80000.hdmi: dw_hdmi_rockchip_select_output color_format: 0

Shivansps, 2023-12-02:
[image]

stevepringle, 2023-12-02:
Hi Joshua, Is there any chance of a prebuilt image using this kernel for experimental purposes?

Joshua-Riek, 2023-12-02:
@stevepringle, you can download the images on GitHub actions. 

Latest build: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7040937947
All mainline builds: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/build-mainline.yml

Joshua-Riek, 2023-12-02:
From my understanding HDMI is very weird right now, some monitors will work, while others will not.

Shivansps, 2023-12-02:
> From my understanding HDMI is very weird right now, some monitors will work, while others will not.

yeah is the same thing im seeing it will work on my USB hdmi capture but not on my monitors except for one.

bagong, 2023-12-02:
Just to make sure that you don't think you're the only one who can do that trick ;-))))

Done on a opi5 feeding to the hdmi-in of a opi5+. It didn't want to display on my ASUS QHD-monitor (haven't tried others yet)

[image]

Joshua-Riek, 2023-12-03:
Anyone try getting panthor working on this? I have not been following the development.

bagong, 2023-12-03:
https://lore.kernel.org/all/20230829160035.1992834b@collabora.com/T/

Joshua-Riek, 2023-12-04:
The patchset for panthor is quite large, so it's hard to move the patches around when the kernel version is bumped so often. 

I would prefer to wait until a smaller patchset is available or a new version, as it will eventually be merged anyway, just takes time.

Joshua-Riek, 2023-12-04:
> Hi.
> I have some experience merging patches in the linux kernel.  That was a while ago but maybe I can do some tests with my opi5+

I just don't want to rebase and merge +200 patches every week. I have a lot of kernel work that needs to be done with the new 6.1.25 BSP kernel. Which a few developers were able to backport panthor to. From my understanding, x11 does not work or display, which is an issue.

Xeue, 2023-12-06:
Just wanted to throw in that I've seen this working on a Rock 5B, it is specific about screen as thought!
Random side question, I'm trying to test gstreamer on this (all works so far!) but I can't find out how to get the mpp264 element for the life of me... I have got mppjpeg and mppdecode. What ppa/package does that come from??

GamingCity, 2023-12-07:
Thats software rendering there is not a single image anywhere that includes the HDMI patches and the panthor driver patches. On top of that it may need a special version of mesa as well, i dont know.

bagong, 2023-12-21:
Here someone has merged panthor into the 6.1 kernel:
https://github.com/hbiyik/linux/tree/linux-6.1-gen-rkr6%2Bpanthor
Thist seems to be 6.7 with panthor:
https://github.com/hbiyik/linux/tree/panthor%2Bmpp%2Brga
And there seems to be mesa 23.3.0 with panthor (branch?, I didn't see a link). In the screenshot Firefox seems to be hardware accelerated ;-)
https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/x-org-drm/1426252-panthor-drm-driver-coming-together-for-newer-arm-mali-gpus

wyf9661, 2023-12-26:
> Here someone has merged panthor into the 6.1 kernel: https://githubfast.com/hbiyik/linux/tree/linux-6.1-gen-rkr6%2Bpanthor This seems to be 6.7 with panthor: https://githubfast.com/hbiyik/linux/tree/panthor%2Bmpp%2Brga And there seems to be mesa 23.3.0 with panthor (branch?, I didn't see a link, maybe this:). https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/26358
> 
> https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/x-org-drm/1426252-panthor-drm-driver-coming-together-for-newer-arm-mali-gpus
> 
> In the screenshot Firefox seems to be hardware accelerated ;-)

I've tried to install this kernel some days ago but hdmi still does not work.
[image]
waiting v6.7 stable version of mainline

wyf9661, 2023-12-26:
also panthor mesa driver below.
[https://gitlab.freedesktop.org/bbrezillon/mesa/-/tree/panfrost-v10?ref_type=heads](url)

GamingCity, 2023-12-26:
VOP is planned for merge in 6.8rc1.
But i heard that panthor does not work with the newer kernel version. So i think they will have to rebase it and adjust.

bagong, 2023-12-26:
> I've tried to install this kernel some days ago but hdmi still does not work.

@wyf9661 I tried the vanilla Mandriva image for rk3588 (6.7 rc4 I think, but no panthor). I thought hdmi didn't work until I plugged it into the middle hdmi port (opi5+), and then it came up in the same way the "server mainline" version here comes up: only on certain monitors. The one that works for me is an quite old ASUS with 1080p/60Hz.

Xeue, 2023-12-26:
I have tried 6.7r4 builds of a few variety and they're all the same. Until VOP is merged in I wouldn't really expect it to work.
The fact it works at all is luck...

Shivansps, 2023-12-26:
Im not actually sure if the VOP2 version that got merged into 6.8rc1 is any diferent than the one Joshua included here, the VOP controller may need just need more work. Or maybe the HDMI element is a separated component.

wyf9661, 2023-12-27:
> @wyf9661 I tried the vanilla Mandriva image for rk3588 (6.7 rc4 I think, but no panthor). I thought hdmi didn't work until I plugged it into the middle hdmi port (opi5+), and then it came up in the [...]
Oh, I usually use dual-screen setup  of my rk3588, I will have a try of either of the hdmi output. And There is also a typec-DP output, I am wondering if this interface works.

wyf9661, 2023-12-27:
> Here someone has merged panthor into the 6.1 kernel: https://githubfast.com/hbiyik/linux/tree/linux-6.1-gen-rkr6%2Bpanthor This seems to be 6.7 with panthor: [...]

I bump to v6.1 and all things goes well. ;-)
[image]

wyf9661, 2023-12-27:
Additionally, I try to use mesa-panthor-driver but when it starts, which is still `llvmpipe` rendering pipeline, so I still use panfork instead of panthor. But the DE runs really more  smoothly.
[image]

bagong, 2023-12-27:
Hi, friendly group. Would someone mind taking my hand by explaining in simple steps (that I could reproduce like a parrot) what to do with u-boot to make it use a newly compiled kernel? Thanks!

wyf9661, 2023-12-28:
> Hi, friendly group. Would someone mind taking my hand by explaining in simple steps (that I could reproduce like a parrot) what to do with u-boot to make it use a newly compiled kernel? Thanks!

FYI
[image]

bagong, 2023-12-28:
Thanks, that's an helpful hint! When I was trying out Mandriva I didn't know what to do about the "fallback". This gives me a path to explore if I understand correctly.

hakotani-o, 2024-01-01:
Happy new year.
@wyf9661 Please check lsmod.

My Opi5
[alarm@alarm ~]$ lsmod
Module                  Size  Used by
joydev                 24576  0
sch_fq_codel           20480  2
fuse                  110592  3
dm_mod                106496  0
nfnetlink              20480  2
ip_tables              28672  0
ipv6                  446464  106
panthor               102400  0
drm_gpuvm              24576  1 panthor
drm_exec               16384  2 drm_gpuvm,panthor
drm_shmem_helper       20480  1 panthor
gpu_sched              32768  1 panthor

stevepringle, 2024-01-01:
Hi Hakatoni-oDid you add panthor yourself, or is it now part of Joshua’s 6.7rc3 build? If you added it yourself, how did you do it?Regards,SteveSent from my iPad

hakotani-o, 2024-01-01:
Hi stevepringle
That is Arch linux for opi.
I use Arch linux AUR package.
 https://aur.archlinux.org/packages/linux-opi
This script can not download linux source.
I change git URL to "https://github.com/hbiyik/linux/tree/linux-6.1-gen-rkr6%2Bpanthor"
and run this script.

Now i try on ubuntu rockchip install 6.1.25 kernel.

wyf9661, 2024-01-01:
> Hi stevepringle That is Arch linux for opi. I use Arch linux AUR package. https://aur.archlinux.org/packages/linux-opi This script can not download linux source. I change git URL to [...]

this package is maintained by me and when I lsmod there is not any driver called panthor, I will change branch tomorrow and have a try.

wyf9661, 2024-01-01:
I've already changed the branch last week, so it is not the reason. I do not upload the mesa driver I build with panthor. Can you tell me you mesa driver branch?
Best wishes, thank you.

hakotani-o, 2024-01-02:
thank you for your reply.
I'm currently trying to build linux-6.1-gen+rkr6+panthor on ubuntu rockchip.
I don't know anything about mesa yet.
Where can I find the mesa source code?
thank you.

Joshua-Riek, 2024-01-02:
Mesa is very broken with the panthor driver, X11 will not work.

wyf9661, 2024-01-02:
> Mesa is very broken with the panthor driver, X11 will not work.

Oh, the default DE in ubuntu is gnome and wayland is set as default, but what I use is kde with x11, maybe it's the reason why it does not work, thanks Joshua.

wyf9661, 2024-01-02:
[image]
I changed to wayland now, I remember that wayland does not work in vendor kernel on v5.10, it is really so great.
Thanks again , Joshua

JFLim1, 2024-01-03:
> I changed to wayland now, I remember that wayland does not work in vendor kernel on v5.10, it is really so great.

Hi @wyf9661,
Is there a pre-built 6.1.25-2-opi, 6.1.25-panthor and mesa-panthor  for download?  Would like to test archlinux with 6.1.25-2-opi, mesa-panthor with KDE Wayland.

wyf9661, 2024-01-03:
> > I changed to wayland now, I remember that wayland does not work in vendor kernel on v5.10, it is really so great.
> 
> Hi @wyf9661, Is there a pre-built 6.1.25-2-opi, 6.1.25-panthor and mesa-panthor for download? Would like to test archlinux with 6.1.25-2-opi, mesa-panthor with KDE Wayland.

no, there is not any official packages for opi5 serials, if you want to use panthor, just change the branch to panthor to have a try.
For my monitor, it does not display with panthor.
Mesa driver of panthor , which @bagong has mentioned before. there is no stable kernel, it is not recommanded to update kernel unless you have way to recover it.

JFLim1, 2024-01-04:
> For my monitor, it does not display with panthor.
> Mesa driver of panthor , which @bagong has mentioned before. there is no stable kernel, it is not recommanded to update kernel unless you have way to recover your environment.

Thanks. Will wait for the kernel 6.1.25 to be officially released.

hakotani-o, 2024-01-06:
Hi @wyf9661
On Arch linux arm , cannot install 6.1.25 kernel (linux-opi)  outside China.
[image]

wyf9661, 2024-01-06:
> Hi @wyf9661 On Arch linux arm , cannot install 6.1.25 kernel (linux-opi) outside China. ![Screenshot from 2024-01-06 09-46-26](https://usercontent.githubfast.com/private-user-images/139629056/294636633-99dbdd85-09a6-41c8-be37-86ba1bf7b4ea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDg4NDgsIm5iZiI6MTcwNDUwODU0OCwicGF0aCI6Ii8xMzk2MjkwNTYvMjk0NjM2NjMzLTk5ZGJkZDg1LTA5YTYtNDFjOC1iZTM3LTg2YmExYmY3YjRlYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNlQwMjM1NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MjVmMGYzMWYyZmU2ZGRhNWIwMTliOWI5MTNkNjBlYTgxNWFiNjAyYTIzOGVlOTgwNzYwNjNiODZkZjgyYzc0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tTsChnIzztnFJfZ_qQWXwdJ23dlzeRhKPe1WDwJjflY)

This aur is a test branch for me. So I use proxy mirror site of github to download source faster. I've already change the url if you'd like to have try. This pkg still has some issues, no sound output, bad gpg, or maybe something I did not discover.

hakotani-o, 2024-01-06:
@wyf9661 thank you.
Git is currently running.

JFLim1, 2024-01-06:
> I remember that wayland does not work in vendor kernel on v5.10,

It seems the new kernel-5.10.198 works with Plasma Wayland session.  Transplanted kernel-5.10.198 from Ubuntu-22.04.3 to Manjaro-Opi5-Plus image, with orangepi firmware, mesa-panfork-git, etc from 7Ji-Repo.  It works.

[image]

wyf9661, 2024-01-07:
> It seems the new kernel-5.10.198 works with Plasma Wayland session. Transplanted kernel-5.10.198 from Ubuntu-22.04.3 to Manjaro-Opi5-Plus image, with orangepi firmware, mesa-panfork-git, etc from [...]

I've changed to kernel that Joshua-Riek build for rockchip, since it is so tired for me to build it myself, 7Ji will bump kernel version  to v6.8 too, maybe you can use this pkg as well.
[image]

JFLim1, 2024-01-07:
> 7Ji will bump kernel version to v6.8 too, maybe you can use this pkg as well.

Thanks for the info.  Looking forward to it provided there is "3D GPU HW Acceleration - Panthor" support.

Looking forward to Joshua Riek kernel 6.1 with panfork & vpu support and mainline kernel with "panthor" support.

JFLim1, 2024-01-13:
> https://aur.archlinux.org/packages/linux-opi

Hi @wyf9661,

I tried to installed this using "yay -S linux-opi linux-opi-headers" but unsuccessul.  It seems to build the packages but would not install.

Any suggestions to rectify this?  Tried a 3 times (each time around 3 hrs).

```
==> Sources are ready.
 -> linux-opi-6.1.43-1 already made -- skipping build
==> Making package: linux-opi 6.1.43-1 (Sat 13 Jan 2024 12:35:06 AM +08)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Sources are ready.
[sudo] password for jfl: 
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (2) linux-opi-6.1.43-1  linux-opi-headers-6.1.43-1

Total Installed Size:  205.07 MiB

:: Proceed with installation? [Y/n] Y
(2/2) checking keys in keyring                                [#################################] 100%
(2/2) checking package integrity                              [#################################] 100%
(2/2) loading package files                                   [#################################] 100%
(2/2) checking for file conflicts                             [#################################] 100%
error: failed to commit transaction (conflicting files)
/usr/lib/modules/6.1.43-1/build/ exists in both 'linux-opi-headers' and 'linux-opi'
Errors occurred, no packages were upgraded.
 -> error installing: [/home/jfl/.cache/yay/linux-opi/linux-opi-headers-6.1.43-1-aarch64.pkg.tar.xz /home/jfl/.cache/yay/linux-opi/linux-opi-6.1.43-1-aarch64.pkg.tar.xz] - exit status 1
[jfl@opi5plus ~]$ 
```

wyf9661, 2024-01-13:
> Hi @wyf9661,
>
> I tried to installed this using "yay -S linux-opi linux-opi-headers" but unsuccessul. It seems to build the packages but would not install.
>
> Any suggestions to rectify this? Tried a 3 [...]

you can just install linux-opi without the other one.

JFLim1, 2024-01-13:
> you can just install linux-opi without the other one.

Rebuild again figure out by just install "linux-opi" alone with:

`sudo pacman -U /home/jfl/.cache/yay/linux-opi/linux-opi-6.1.43-1-aarch64.pkg.tar.xz`

linux-opi-6.1.43-1-aarch64.pkg.tar.xz installed successfully.

`sudo pacman -U /home/jfl/.cache/yay/linux-opi/linux-opi-headers-6.1.43-1-aarch64.pkg.tar.xz `

But "linux-opi-headers" failed to installed.

The resultant "linux-opi-6.1.43-1" booted up successfully but ONLY in X11 session eventhough "Gnome" session (which should be Wayland) is selected.

Did you have the same experience only X11 is available with linux-opi-6.1.43-1?

wyf9661, 2024-01-14:
> > > > https://aur.archlinux.org/packages/linux-opi
> > > 
> > > 
> > > Hi @wyf9661,
> > > I tried to installed this using "yay -S linux-opi linux-opi-headers" but unsuccessul. It seems to build the packages but would not install.
> > > Any suggestions to rectify this? Tried a 3 times (each time around 3 hrs).
> > > ```
> > > ==> Sources are ready.
> > >  -> linux-opi-6.1.43-1 already made -- skipping build
> > > ==> Making package: linux-opi 6.1.43-1 (Sat 13 Jan 2024 12:35:06 AM +08)
> > > ==> Checking runtime dependencies...
> > > ==> Checking buildtime dependencies...
> > > ==> WARNING: Using existing $srcdir/ tree
> > > ==> Sources are ready.
> > > [sudo] password for jfl: 
> > > loading packages...
> > > resolving dependencies...
> > > looking for conflicting packages...
> > > 
> > > Packages (2) linux-opi-6.1.43-1  linux-opi-headers-6.1.43-1
> > > 
> > > Total Installed Size:  205.07 MiB
> > > 
> > > :: Proceed with installation? [Y/n] Y
> > > (2/2) checking keys in keyring                                [#################################] 100%
> > > (2/2) checking package integrity                              [#################################] 100%
> > > (2/2) loading package files                                   [#################################] 100%
> > > (2/2) checking for file conflicts                             [#################################] 100%
> > > error: failed to commit transaction (conflicting files)
> > > /usr/lib/modules/6.1.43-1/build/ exists in both 'linux-opi-headers' and 'linux-opi'
> > > Errors occurred, no packages were upgraded.
> > >  -> error installing: [/home/jfl/.cache/yay/linux-opi/linux-opi-headers-6.1.43-1-aarch64.pkg.tar.xz /home/jfl/.cache/yay/linux-opi/linux-opi-6.1.43-1-aarch64.pkg.tar.xz] - exit status 1
> > > [jfl@opi5plus ~]$ 
> > > ```
> > 
> > 
> > you can just install linux-opi without the other one.
> 
> Rebuild again figure out by just install "linux-opi" alone with:
> 
> `sudo pacman -U /home/jfl/.cache/yay/linux-opi/linux-opi-6.1.43-1-aarch64.pkg.tar.xz`
> 
> linux-opi-6.1.43-1-aarch64.pkg.tar.xz installed successfully.
> 
> `sudo pacman -U /home/jfl/.cache/yay/linux-opi/linux-opi-headers-6.1.43-1-aarch64.pkg.tar.xz `
> 
> But "linux-opi-headers" failed to installed.
> 
> The resultant "linux-opi-6.1.43-1" booted up successfully but ONLY in X11 session eventhough "Gnome" session (which should be Wayland) is selected.
> 
> Did you have the same experience only X11 is available with linux-opi-6.1.43-1? My experience is "Wayland" session works smoother on OPI5-Plus with mesa-panfork so far.

I'm sorry, the upstream repo is not maintained by me and mostly it is an experimental branch of 6.1, which can not run stably with some problem. Now I am aiming to the mainline kernel of 6.7 and don't want to spend too much time on it. If you need a stable runtime environment, you'd better select the vendor kernel.

JFLim1, 2024-01-15:
I thought the kernel-6.1.43 was the rockchip bsp kernel.

Both kernel-5.10.160 and kernel-5.10.198 is stable so far and kernel 5.10.198 support KDE Wayland.

HSTe, 2024-01-25:
It looks like collabora has just added panthor to a public branch based on 6.8-rc1, but it's only added on the rock5b and and evb1 board. So looks like the hdmi and gpu parts must be added to other boards dts and its still wip. 

https://gitlab.collabora.com/hardware-enablement/rockchip-3588/linux

If you want to follow the development for the rk3588 look at:
https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md
https://lore.kernel.org/all/?q=rk3588
https://lore.kernel.org/all/?q=panthor

Joshua-Riek, 2024-01-25:
I saw that, but I've been trying to re-work the entire boot process, so I've not had much time for kernel things.

My current implementation has problems when it comes to installing multiple kernels. Other implementations also have flaws, so I'm pretty much exploring anything at this point.

HSTe, 2024-01-25:
Are looking to bootprocess like using https://github.com/edk2-porting/edk2-rk3588 ?

Joshua-Riek, 2024-01-25:
No, it introduces a few problems and will not work with device tree overlays. I'm talking about extlinux / u-boot-menu or flash kernel. But both have problems as well.

7Ji, 2024-01-28:
FYI, you could store dtbs of different kernels under different subfolders, to have a tree structure like the following: 
```
/boot
├── booster-linux-aarch64-7ji.img
├── booster-linux-rkbsp-joshua-git.img
├── dtbs
│   ├── linux-aarch64-7ji
│   │   ├── amlogic
│   │   └── rockchip
│   └── linux-rkbsp-joshua-git
│       └── rockchip
├── extlinux
│   └── extlinux.conf
├── vmlinuz-linux-aarch64-7ji
└── vmlinuz-linux-rkbsp-joshua-git
```

This way kernels, initrds and dtbs of multiple kernel packages won't conflict with each other.

And then use simply extlinux.conf to configure a boot menu entry for each kernel:

```
MENU TITLE Select the kernel to boot
TIMEOUT 30
DEFAULT linux-aarch64-7ji
LABEL   linux-aarch64-7ji
        LINUX   /vmlinuz-linux-aarch64-7ji
        INITRD  /booster-linux-aarch64-7ji.img
        FDT     /dtbs/linux-aarch64-7ji/rockchip/rk3588s-orangepi-5.dtb
        APPEND  root=UUID=0f113376-5f2c-4f96-ac28-99ba3b39673e rootflags=subvol=@,compress=zstd:3 rw
LABEL   linux-rkbsp-joshua-git
        LINUX   /vmlinuz-linux-rkbsp-joshua-git
        INITRD  /booster-linux-rkbsp-joshua-git.img
        FDT     /dtbs/linux-rkbsp-joshua-git/rockchip/rk3588s-orangepi-5.dtb
        APPEND  root=UUID=0f113376-5f2c-4f96-ac28-99ba3b39673e rootflags=subvol=@,compress=zstd:3 rw
```
Each kernel can have their different boot options, including overlays. (`FDTOVERLAYS`, I'm not currently using any, though)

U-boot would prepare a menu automatically, on both serial console, and potentially hdmi if driver is built-in. Like how you would use a bootloader to choose entry on x86_64.

This does break some assumptions many distros made about the kernel and dtbs layout. And many just stick always to the way it was used to be from early rpi days.

For me, I started to use this setup in 2022 when creating my amlogic-s9xxx-archlinuxarm project because I don't need to care about "existing" setups and wanted to have a multi-kernel setups like on x86_64. If you do take this transition for a project that's widely used then it sure would be a painful transition...

Joshua-Riek, 2024-01-29:
> FYI, you could store dtbs of different kernels under different subfolders, to have a tree structure like the following:
>
> ```
> /boot
> ├── booster-linux-aarch64-7ji.img
> ├── [...]

I do like this method, but id need to backport the u-boot-menu package to Jammy as it does not support device tree overlays. For now, I'm going to use flash-kernel like the Raspberry Pi 4, it's messy and clutters the boot partition. But maybe when I work on Ubuntu 24.04 images I'll switch to extlinux.

GamingCity, 2024-02-01:
Panthor is marked as ready/sent on collabora rk3588 mainline status, except for the rockpi for some reason. This means it will be merged next i guess

GamingCity, 2024-02-16:
OK HDMI-PHY is finally merged.
https://lore.kernel.org/all/170808468224.352706.3107795890372449699.b4-ty@kernel.org/

I think panthor is just about to be merged too.

Joshua-Riek, 2024-03-02:
nice :sunglasses:

closed 2024-12-16 by github-actions[bot]