Joshua-Riek/ubuntu-rockchip #499: Trouble Updating (forbidden IP)

uglyfish333, 2023-12-04:
So I installed the OS to the emmc of my 5b a number of months ago, and basically through it in the drawer for a while until recently I finally decided to use it as a dedicated home assistant server. Since there's been a few revisions since I installed, I figured I should upgrade to the latest version for better stability control and I believe I read better bluetooth support.

The wiki mentions it should be updatable via apt, so I tried running 'sudo apt update' and got a bunch of errors stating the IP was forbidden.

`Hit:1 https://download.docker.com/linux/ubuntu jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy InRelease                     
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease             
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:5 https://ppa.launchpadcontent.net/jjriek/orangepi5/ubuntu jammy InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Err:7 https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu jammy InRelease
  403  Forbidden [IP: 185.125.190.80 443]
Err:8 https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu jammy InRelease
  403  Forbidden [IP: 185.125.190.80 443]
Reading package lists... Done
E: Failed to fetch https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.80 443]
E: The repository 'https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.80 443]
E: The repository 'https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

Is this a bug or am I just going about updating wrong?

uglyfish333, 2023-12-04:
uname -r gives the following, in case current version is important:

5.10.110-orangepi-rk3588

Joshua-Riek, 2023-12-04:
I'm sorry, but you are on a very old version that no longer is supported. You will need to re install your system with the latest available version.

uglyfish333, 2023-12-04:
Figured that might be the case. Once I reinstall to the latest version will future updates be available without complete re-install?

By the way, really appreciate all your hard work, and to be on here responding to everyone's questions so quick is pretty awesome.

Joshua-Riek, 2023-12-04:
I made a mistake with the kernel and packaging it to distribute updates early in the project. I had no idea what I was doing, to be honest, but now I've refined the process and have a good handle on an update distribution process.

But yes, you will receive system and future updates via apt when using the latest release.

closed 2023-12-04 by Joshua-Riek