Joshua-Riek/ubuntu-rockchip #502: Ubuntu server kernel panic on Orange Pi 5 plus

fcastello, 2023-12-05:
I have an Orange Pi 5 Plus with a 1TB nvme drive. I installed the presintalled server version so only CLI.

I also didn't install anything on it yet other than upgrading existing packages.

I am booting from nvme drive

I left it overnight and when I tried to connect to it I saw a kernel panic on the hdmi.

I had to unplug and plug it back in to get it to reboot.

Also I am experiencing kernel panics some times at boot which get fixed but unplugging and plugging back again.

I am attaching full system log and let me know  what else I can do to help try fix this issue.
[sys.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13572723/sys.log)

Joshua-Riek, 2023-12-05:
You may not have an adequate power supply. Are you using a 5V 4A power supply?

The Orange Pi 5 series has been thoroughly tested and should not have this sort of issue, but I will spin up a new image and see if I can reproduce this.

Joshua-Riek, 2023-12-05:
I just tested on my board and with 5 reboots after a system update I see no kernel panics. I can leave my board on overnight to observe.

fcastello, 2023-12-05:
I am using the official power supply from their website 5V 4A

fcastello, 2023-12-05:
So far I haven't seen crashes using the Orange pi ubuntu build if that helps in any way. However I like a more vanilla ubuntu closer to the images ubuntu provides for raspberry pi and this is the closest I found.
I will conduct also a test with Armbian Jammy version which is based of Ubuntu Jammy.

fcastello, 2023-12-05:
There are some things I still need to test like flashing this image to SD and boot from there. I am suspecting the boot form nvme has something to do with it. Will keep the issue updated with the latest news

Joshua-Riek, 2023-12-05:
How are you booting from the nvme if you never booted from an SD card? You may be using some other 3rd party bootloader that causes issues.

fcastello, 2023-12-06:
I booted from SD with the ubuntu provided by the orange pi website first and then flashed the bootloader to spi. Forgot to mention that.
Also I have been running Armbian Jammy version since last night with kernel 5.10.160 and hasn't crashed yet. Which seems to have similar setup like your image. 

I will give another try, and also test the 6.6.0 Kernel version and keep you updated

fcastello, 2023-12-06:
OK I got the exact same issue on Armbian. It just crashed in the same way with a kernel panic

I will try the 6.6.0 version. Also I am booting from eMMC now to rule out the issue is with the nvme.

fcastello, 2023-12-06:
Let me know if there is anything I can collect for more details debugging

fcastello, 2023-12-06:
I can't reproduce the kernel panic consistently, they happen randomly. So I have a couple of tests more to run.
Like run it on the sd card, on eMMC and nvme drive.
The nvme drive is a brand new Samsung MZ-V8V1T0 which I got for the good reviews, but still doesn't mean it's not bad.

When You say `should not be using the spi image from orangepi` what shoud I be using to boot and flash the SPI bootloader

I didn't understand a bit what this is `the spl overwrites everything else in the boot process. Additionly different spl and bl32 blops (which would be your case) can cause memory training issues leading to issues.` 

If you have some links where I can refer to for setting up the right bootloader I would really appreciate it. Also I will keep trying to  consistently reproduce the kernel panic

fcastello, 2023-12-08:
I have tried so many things that now I can't boot this images from SD, which I want to do to load the correct bootloader to the spi.

On this docs https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5-Plus I am looking inside the image and cant find the image file for the bootloader specially to run this ```sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
sync```

/lib/u-boot-orangepi-rk3588/rkspi_loader.img doesn't exist in the image. Should I build the image or something?

fcastello, 2023-12-08:
i just found the image and it's on /lib/u-boot/rkspi_loader.img. Turned out I mounted it on the orange pi ubuntu flavor and since /lib/u-boot-orangepi-rk3588/ it's a simlink it was pointing to somewhere else because I mounted the image partition to /mnt. It was my bad
I will do some more tests and keep you posted

fcastello, 2023-12-08:
I have changed bootloader and booted again with the stable version. I will leave overnight and see if I get the spontaneous kernel panic.

Something I noticed. Which is probably a different issue, let me know if I need to create a different issue
[image]

In the screenshot you can see that the kernel process `kworker/4:1+events` it's always at the top, it's not hogging the CPU but seems weird to me. Also notice the state `D` which googling a bit shows Uninterruptible sleep (usually IO). And if we look at the load average it flats at 1.0. And the system is vanilla, I am not running anything new other than services that are already shipped with the image. Since it's not a busy system that is concerning to me as load average should be close to 0 unless I start running stuff.

Any help on that regard is welcome. Also if you need me to create a different issue I could (I think this is not related to the crashes I experience)

Also if it helps I see the same behavior booting from SD, eMMC or nvme. As well as booting armbian image but I don't see the same behavior with the orange pi provided image

fcastello, 2023-12-08:
I moved to booting the mainline 6.6.0 version and I booted correctly now.

The issue mentioned in the previous message is not happening on mainline 6.6.0

The only issue I see is that I got no HDMI output, but I was able to log in with SSH on server version.

I will leave overnight and see if the kernel has any crash

fcastello, 2023-12-11:
6.6.0 Kernel has not crashed yet and I left it the whole weekend. However there are some issues, like no HDMI output, The pwm fan is not working as well. Of course I know is an experimental image.
I will give another try to the stable image

fcastello, 2023-12-14:
Good news that I didn't get a crash on kernel 6.6.0, still didin't find a way to solve the hdmi output or the pwm fan.

I see there is a new stable version form 5 days ago. I will try that one and see if I get crashes with that one

Joshua-Riek, 2023-12-14:
Sorry for not following up sooner, ive been very busy with work. HDMI should be somewhat working on 6.7-rc4 that I have available on GitHub Actions (link below).

I have some misc reports of kernel panics from individuals in Discord, so I will be trying to dive into this a little bit more.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7081798831

fcastello, 2023-12-14:
No worries I understand this is best effort. I haven't been able to spend time on it the last couple of days either.

Joshua-Riek, 2023-12-14:
Could you try this image? I updated the DDR and BL31 blobs. This is responsible for memory training during the boot process, if this is a memory issue it should help solve the problem.

When you boot the image, please install the bootloader with the below command, it may take 5 minutes to finish **(If you previously installed the bootloader to the SPI this is required for the updated DDR and BL31 blobs to take effect, this is applicable even if booting from an SD card)**.
```
sudo u-boot-install-mtd /dev/mtdblock0
```
Also make sure that no eMMC module is connected, after installing the bootloader, please reboot the system, and the updated DDR and BL31 blobs should be in effect. 

> Don't issue an apt update before installing the bootloader to the SPI or after the package updates. The build below is more of a Debug build without packaged changes. (After installing the bootloader to SPI it's fine to update, but again do not update the bootloader again using the above u-boot-install-mtd command).

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7187540524

rshramko, 2023-12-16:
Hi @Joshua-Riek 

I received my board a couple of days ago and I seemed to have the same issue with kernel panic like @fcastello. The image from https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7187540524 works pretty well for me.

Joshua-Riek, 2023-12-16:
> Hi @Joshua-Riek 
> 
> I received my board a couple of days ago and I seemed to have the same issue with kernel panic like @fcastello. The image from https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7187540524 works pretty well for me. 

Thank you very much for this information! I will take a look at pushing a bootloader update. But if we can get more users to confirm this fixes the issue, that would be great to know this is a proper solution.

fcastello, 2023-12-28:
@Joshua-Riek I have been running the image from https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7187540524  and haven't experienced any kernel panics for a few days. So I think this issue is solved.

fcastello, 2023-12-28:
I am still experiencing this issue https://github.com/Joshua-Riek/ubuntu-rockchip/issues/502#issuecomment-1847592545

I will open another issue for that one which is not related. Also I am experiencing that on Armbian on the same kernel version. 
But I did not experience that issue with the 6.6.0 kernel.

Thanks for the help

Joshua-Riek, 2023-12-28:
> I am still experiencing this issue https://github.com/Joshua-Riek/ubuntu-rockchip/issues/502#issuecomment-1847592545
>
> I will open another issue for that one which is not related. Also I am [...]

I assume this is on the bsp kernel. If so, it's likely to nerver be solved. I think the kernel will be abandoned once the 6.1 BSP comes out, unless it turns out to be even more of a mess than the current 5.10 BSP.

fcastello, 2024-01-03:
Yes I am on the 5.10 BSP kernel and still experiencing it. Weird I did not experience that on the orange pi provided image which is also running kernel 5.10. But I don't like the officially provided image as it doesn't use the official ubuntu repositories. So I am looking for a stable solution

Joshua-Riek, 2024-01-03:
Can you provide uart logs during the boot process? I would like to see if there is a mismatched spl and ddr blob problem going on. 

I saw someone on the Armbian forums have a very similar issue, and the blob mismatch can cause RAM stability issues.

fcastello, 2024-01-03:
I will try to get the logs. And I did experience the same thing on Armbian with 5.10 kernel.
I am trying Armbian with 6.7-rc1 kernel and issue seems to be gone

greenplugin, 2024-01-12:
Hi guys I got same issue on kernel from latest release, but kernel 5.10.160 works for me. And kernel 6.7.0 also works good, but it looks that on 6.7.0 gpu support is not works properly. 
If it can help to fix the issue I can flash image from latest release and attach it here.

Joshua-Riek, 2024-01-12:
> Hi guys I got same issue on kernel from latest release, but kernel 5.10.160 works for me. And kernel 6.7.0 also works good,

Your statement does not make sense. The latest release uses 5.10.160 which you say works. 

> it looks that on 6.7.0 gpu support is not works properly.

I'm getting so tired of explaining this over and over. Mainline Linux is **highly experimental** don't expect everything to work because the hardware is not fully supported. Do some research before complaining.

greenplugin, 2024-01-12:
Yes, about 6.7 I understand. 
I was generally surprised that it started on the first try and without major problems
About release it's strange I tried release and got panics? then I found this thread and tried few versions from pipelines and one of them on kernel 5.10.160 works good. I'll download latest release again and re-check.

Joshua-Riek, 2024-01-12:
If you have installed the bootloader to the spi flash from another system, it could cause memory instability, which would cause a kernel panic. I'm sure this is the problem here.

greenplugin, 2024-01-12:
About bootloader after I got the board I triied officeial images and debian, opios didn't work for me, then I tried unutu xfce from orangepi site and it was work. But official images are glitchy includes their ubuntu and works not very good. And then I tried to find another more clean os. I tried latest armbian and got kernel painc and your ubuntu build and got same behavior and then I tried one by one builds from pipelines and one of them works for me. 

btw, I have a question about boot on spi flash. 
if I boot from an SD card, does the SPI somehow participate in the boot? I thought that if there is a bootable SD card, then boot occurs directly from it.
And if on spi written boot from any release with kernel 5.10.160 should work any other build of this ubuntu with same kernel or not?

greenplugin, 2024-01-12:
right now I have flashed latest release to sdcard and got kernel panic again 
[uart_01.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13922153/uart_01.log)
so I'll try to find again that image what works for me and let you know what pipeline is it.

Joshua-Riek, 2024-01-12:
If you boot from sd card the SPL from the SPI will be used, it can cause all sorts of trouble. What power supply are you using?

> right now I have flashed latest release to sdcard and got kernel panic again
[uart_01.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13922153/uart_01.log)
so I'll try to find again that image what works for me and let you know what pipeline is it.
Or maybe I can found build number in the image?

I need the full u-boot output to determine if SPL is the issue.

Joshua-Riek, 2024-01-12:
> [   44.346621] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008

This is the panic message which again makes me believe this is the SPL and DDR blob not matching leading to a memory issue.

greenplugin, 2024-01-12:
Hmm I got another error and kernel panic during loading (I never got this error before) https://gist.github.com/greenplugin/abf4ba8f831f0c5c66fd0db90aac6bf3  
And on the second try I got same behavior as before https://gist.github.com/greenplugin/40cb160f22b5849eee15d5bb4562617e. 
I tried to use differnt power sources, offcial power supply, xiaomi power supply 5v3a, laboratory power supply 5v 5a. Ang got same behavior on all of them. Latest tests on power supply that 5v 3a. But one of images from builds.

greenplugin, 2024-01-12:
As I remember the image from this post https://github.com/Joshua-Riek/ubuntu-rockchip/issues/502#issuecomment-1856012729 works good for me. But I'll recheck. Or I can give logs to you from image that works.

Joshua-Riek, 2024-01-12:
Then give your system a reboot, and you will no longer have an issue.

greenplugin, 2024-01-12:
Thanks. I have insert sd with working image and erased mtdblock0 using dd. And then insert sdcard with latest release and again got kernel panic. then found in logs 
> ERROR: loader&trust unmatch!!! Please update trust if need enable dmc
again :thinking:. 

And then I remembered that I had not disconnected the emmc, turned off the board, disconnected it and I could successfully boot into the working system.

Thank you very much for your help. It looks like my issue has not been related to this thread.

Joshua-Riek, 2024-01-12:
@fcastello I dont recall if you ever checked the u-boot logs, or posted them. It would prove that you have a SPL and DDR mismatch leading to memory instability, like the above user. 

I will likely close this issue as resolved as you both have the exact same kernel panic messages.

closed 2024-01-28 by Joshua-Riek