Joshua-Riek/ubuntu-rockchip #505: rock5a how to switch to Wayland?

eternal-Eason, 2023-12-09:
Hello! My board is rock5a, burning the image (Ubuntu-22.04.3-preinstalled-Desktop-Ar64-Rock-5A) found that using the browser chromium, choppy animation, now using x11, how to switch to Wayland? thank you.

eternal-Eason, 2023-12-09:
[image]
Error message when first boot

eternal-Eason, 2023-12-09:
How to debug this problem？

Joshua-Riek, 2023-12-09:
X11 is very slow and choppy, Wayland is much better and the only supported environment at the moment.

You should try to re-install because the setup install failed.

bagong, 2023-12-09:
But Chromium doesn't run in actuall Wayland, right? Or if one switches to it, hardware-support get's lost, right?

eternal-Eason, 2023-12-11:
> X11 is very slow and choppy, Wayland is much better and the only supported environment at the moment.
> 
> You should try to re-install because the setup install failed.

Thank you for your response. The root file system I compiled had some issues, but after recompiling it, I won't encounter those errors again. Thank you.

closed 2023-12-11 by eternal-Eason