Joshua-Riek/ubuntu-rockchip #507: rk3588 adaptation

liuzhe1947, 2023-12-11:
HI Joshua-Riek

I have a rk3588 motherboard that was originally designed with the Android 12 system. I have extracted the device tree. Could you please adapt it to the Ubuntu system? Thank you.

This is the device tree file I extracted. 

1.https://github.com/liuzhe1947/Rk3588-Information
2.https://wwmh.lanzoul.com/iQH471hew1fg 密码:5rd8

Joshua-Riek, 2023-12-13:
I don't want to reverse-engineer an Android device tree for a random RK3588 board. This will take way too much time and effort.

closed 2023-12-13 by Joshua-Riek

liuzhe1947, 2023-12-14:
Sorry, after my testing, this is a universal device tree. I have replaced the Armbian device tree with my own, and there is no problem booting it up.