Joshua-Riek/ubuntu-rockchip #510: About OpenGL 3.3 on the 1.31 image

Shivansps, 2023-12-13:
I just tested that image and glxinfo reported 3.3 opengl version and it was not just a string override has im very sure that geo shaders were working (3.2 feature). 

But i have to compile panfork from source because with the game i trying i get a IEQ exception crash due to some bug in the mesa optimisations. In sorth i need to compile my own version because i need to disable one optimisation that affects bitfrost-valhall.

But after doing this i was getting opengl 3.1 and geo shaders were not working. How do you archive full 3.3 support?

Joshua-Riek, 2023-12-13:
GPU accel is provided by Panfork, which is a massive usperspace hack, we have no alternative right now.

GamingCity, 2023-12-15:
Yes i know that. That i meant that there seems to be two versions of panfork or there is a configuration switch im not aware of.

The panfork included in the current non-mainline image reports OpenGL 3.3 on glxinfo and it seems to be actually be real 3.3, as at least geo shaders (3.2 feature) seems to work. So is not just some string override.

Panfrost on mainline Mesa is 3.1 and when i compile panfork from this source https://gitlab.com/panfork/mesa (as i mention it i need to disable one optimization), i get 3.1 and geo shaders dosent work, so it is in line with what is present in panfrost on Mesa.

But as i said the panfork version you included seems to be newer than that or im missing something here.
(yes i posted with the wrong account, sorry)

Joshua-Riek, 2023-12-15:
Panfork is a complete hack, I would recommend not spending much time on it. I use the ppa created by amazingfate which packages panfork.

closed 2023-12-21 by Joshua-Riek