Joshua-Riek/ubuntu-rockchip #512: OrangePi 5 Module WiFi dont Work Ubuntu server

WINTX, 2023-12-14:
I have an orange pi 5 with its nvme Wifi module but it doesn't work, if I type iwconfig nothing appears, I don't know if I'm doing something wrong. I use Ubuntu server and the latest version

thnx!

[image]

closed 2024-01-12 by Joshua-Riek

dsburns, 2024-01-25:
@Joshua-Riek - Can confirm the same issue on my board.  I'm using the wifi module found here: https://a.co/d/52SdOc6

Does this require a specific overlay?

Joshua-Riek, 2024-01-25:
it will work, does not need an overlay its pcie

dsburns, 2024-01-25:
Sorry, I just realized.  I'm running an Orange PI 5 Plus.  I've tried two different modules and the OS doesn't recognize either.

Joshua-Riek, 2024-01-25:
This gives me no information?