Joshua-Riek/ubuntu-rockchip #516: Orange Pi 800

davidstoll, 2023-12-16:
Sorry if this has been asked, but I looked and didn't see anything.

Can I request support for the Orange Pi 800?  It is basically an Orange Pi 5, but in a keyboard form factor (like raspberry pi 4 and 400).  I'm sure there are differences and I'm probably oversimplifying.  Maybe another existing image would be compatible?

In any case, if it is not yet supported, can I please humbly request support for it?

Thank you!

bagong, 2023-12-18:
Of course it's Joshua's prerogative to answer, but it sounds very unlikely to me. The official OS releases for the opi800 might appear similar to the opi5, but both computers have different processors: the opi5 a RK3588(s), and the opi800 a RK3399. The opi800 might be a bit more powerful than the rpi400, but basically the opi800 orients on the rpi400 (to put it politely ;-) ). As to support by Ubuntu Rockchip, the "About" says "Ubuntu 22.04 for Rockchip RK3588 Devices", which excludes the opi800.

Joshua-Riek, 2023-12-21:
Sorry but no, its old hardware at this point.

closed 2023-12-21 by Joshua-Riek