Joshua-Riek/ubuntu-rockchip #519: No IP assignment to Ethernet

somu1795, 2023-12-18:
Hi, First of all, thanks a lot for these wonderful kernel and builds.

**Device**: NanoPi R6C
**Issue**: This device has 2 ethernet ports(1G and 2.5G), upon connecting to any router using the 1G port, this device does not assign any IP address, however if I switch to 2.5G port it immediately gets the IP address. 

I have tried with couple of routers(Mikrotik, TP-link, D-link etc)

Kindly let me know if this is  a known issue? or if any further info is required.

Joshua-Riek, 2023-12-19:
I'm aware of this issue, I also encountered it with Mainline Linux. Needs some additional debugging and research to figure out the exact cause.

somu1795, 2024-01-14:
Just to add more, sometimes the 2.5G (which usually works fine) also disconnects for couple seconds and connects back again.
This happens more when there's network activity (streaming or transferring files etc. to/from the device).

I have verified that this is not a power supply issue, since I can run benchmarks for long time without any disconnection.

abacon118, 2024-05-25:
I've noticed something similar on the Orange Pi 3B.  I plugged up a USB C dock with ethernet and it will only connect over that.
I've not tried with official firmware yet.

wgfilho, 2024-07-03:
Something happened to me on an Orange Pi 3B. I plugged a usb-ethernet adapter and it worked, but the onboard ethernet does not work. It never gets ip, or pings anyone even if I set the ip manually. 

I also didn't try the official firmware, but I plan to do that later today.

navarro967, 2024-07-04:
I'm also running into this on a FriendlyElec CM3588-NAS but on the RTL8125B(2.5G) device. Unfortunately the only USB NIC I have is also a RTL8125B.

The onboard NIC is using the r8169 driver however the USB NIC is using cdc_ncm.

From what I can tell it is able to negotiate the correct link speed depending on which switch I connect it to however I don't see DHCP handshake occurring nor am I able to ping my gateway with static IP configuration. I did notice the onboard ethernet works when using the official image. I'll have to check what driver the official FriendlyElec image is using though I'm not entirely keen on running images that came from a Google Drive with a bunch of software baked-in.

Joshua-Riek, 2024-07-04:
FriendlyElec CM3588-NAS is not supported, anyone is welcome to make a pull request and / or donate the hardware, I can not feasible spend money obtaining all these devices.

navarro967, 2024-07-04:
Hey there Josh,

I am aware of it not being supported by this project yet I only wanted to
share and document. I am proficient in Linux and fully intend to contribute
back any changes required to have the device supported if possible.

Thank you for all of your hard work.

wgfilho, 2024-07-06:
Hi! I tested with Orange pi's debian image and the network card works fine. Is there anyway I can help you debug it? 

Thanks!

nemanjam, 2024-10-23:
And what is the solution? `io` command isnt defined in Ubuntu and can not be installed.

``` bash
 io -4 0xFDC60284 0x3f3f3f3f

io -4 0xFDC6028C 0x003f003f

io -4 0xFDC60298 0x3f003f00

io -4 0xFDC6029C 0x3f3f3f3f

io -4 0xfdc60388 0xFFFF0049 
```

closed 2025-02-17 by github-actions[bot]