Joshua-Riek/ubuntu-rockchip #520: opi5+ pcie-wifi-bt cartd from op rtl8852be shows up twice in Gnome and `ip link show`

bagong, 2023-12-18:
Hi,

today I got a new 32GB-ram opi5+ (rev 1.3.1) with a Realtek pcie-network card branded by opi (the kit was bought from Xulong via Amazon US (but sent from China)). I assembled it and copied the latest 1.31 release image to an nvme disk. Then I booted (without an SD card inserted) and was surprised to find that the system found the nvme-disk and booted from it (no need to specify anywhere that I was wanting to use the nvme-drive as boot drive, it just worked).

Basically everything seems to be fine: all drives recognized, ethernet etc. Wifi had already been recognized at first boot, but with an oddity: at the first install wizard it offered me my home-network twice - once by it's name, and a second network entry under the heading Realtek, and then again that same name of my home wifi. I chose one of them and proceeded. After the wizard was finished the wifi-network did not work instantly, but I could log into the wifi-network via Gnome settings. Then I rebooted, and on second boot the network login worked automatically. So in principle it works fine.

However, Gnome's system panel now shows two wifi-networks (if I am not logged in, one displays as PCI-Wi-Fi and the other just as Wi-Fi).

`ip link show` also displays two wireless devices.  One device is called `wlP2p33s0` (address 40:9c:XX:XX:XX:XX), and the second one `wlan1` (42:9c:XX...). In the Gnome settings/Wifi I get two entries, both called `Realtek Wi-Fi`. The first one shows `connected` and has the properties gear with the expected entries (WPA3, 2.4/5GHz, Hardware address. The second one lists the network, is however not connected, and cannot be connected either. In `ip link show` the status for that second device (wlan1) is given as DOWN.

`ls /sys/class/net` returns `enP3p49s0  enP4p65s0  lo  wlan1  wlP2p33s0`, so again the two wireless devices.

`lspci -nnk` however only returns one (wireless) Network controller (and two ethernet controllers):

````
0002:21:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
	Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
	Kernel driver in use: rtl8852be
	Kernel modules: 8852be
````

Bluetooth works although it took 2 or three boots (and maybe a system update) until it remembered that it was supposed to stay on.

It might just be a cosmetical issue, as everything seems to be working fine. However I'd like to get rid of that second device, but have no idea how ;-) (and I wonder why it happened...)

Best
.b.

bagong, 2023-12-18:
I've done another install. This time I took the image from "Actions" and installed it onto a SD-card. Also I didn't use wifi, but ethernet for the first boot and setup wizard.

With respect to Wifi the result is exactly identical. Then I forced a login for both "wifi-cards", and in fact they do get different IPs - so this is more than cosmetical. Another observation: after each boot, the system seems to create another network profile.

After the setup-wizard I had two network profiles: homenetwork and homenetwork 1. After the first boot I got antoher network profile: homenetwork 2. The system had only logged inot homenetwork 1, but homenetwork was still present. It appears that after each boot another network profile is added.

Cheers
.b.

ditstyle-yang, 2023-12-19:
Use the rtw-89-dkms package from radxa-repo and disable (blacklist) the original 8852be module in the kernel, or compile one from https://github.com/lwfinger/rtw89 following the tutorial.
I've been using rtw89-dkms for quite a while now and haven't found any problems so far.

bagong, 2023-12-19:
Thank you @ditstyle-yang . Are you implying you've seen this behavior before? I've got multiple rpi5s, and this is the first time I observe this behavior. The main difference compared to other installs is that this is a opi-branded wifi-card. In all other respects there seems to be no difference, so I am kind of puzzled what is actually going on. 

Both devices seem to use the same kernel module (pci:rtl8852be). Wouldn't blacklisting one cause both devices to disappear? Ah, but you mean the newly added rtw89 module would create another device without duplication? (unfortunately I didn't write down the exact name of the network card, and I am hesitant to disassemble my kit (passive cooling with thermal pads), in case someone comes and says: oh, it's a known issue, simply do this and that...)

ditstyle-yang, 2023-12-19:
With my rock-5b and other brands of rtl-8852be pci modules, all the used image from radxa, armbian and this repo had the problem you mentioned until I replaced the driver with rtw89 to solve it.
In fact, all these mirrors mentioned above basically use the 8852be-dkms package from the radxa-repo, and it is this package that has this problem.
The rtw89-dkms driver doesn't have this problem, it only provides a network interface, it doesn't have the problem that the network changes after a system reboot, and the bluetooth works fine, but it should be noted that this driver requires firmware-realtek >= 20230117-1 or linux-firmware >= 20220329. git681281e4-0ubuntu3.10 and may require manually extracting the zstd compressed firmware.

bagong, 2023-12-19:
Okay, I'll experiment with that. I have however come closer to understanding what is going on: I have 3 wifi receivers in my home, and their radius overlaps somewhat. Two are connected in a fritzbox wifi mesh, and the third one is a receiver connected to my main router via cable. All three use the same network name and credentials. I can see now that the two wifi devices connect to different receivers in my home network. So rather than showing multiple networks for a single network card, two devices are created for different receivers (I say receiver in lack of a better word. One is a fritzbox wifi receiver, and the other is a devolo powerline router that creates its own network. And the main router creates its wifi network too).

bagong, 2023-12-19:
@ditstyle-yang , sorry to ask again, but said repo makes this warning: "This code will build on any kernel 5.6 and newer as long as the distro has not modified any of the kernel APIs. IF YOU RUN UBUNTU, YOU CAN BE ASSURED THAT THE APIs HAVE CHANGED. NO, I WILL NOT MODIFY THE SOURCE FOR YOU. YOU ARE ON YOUR OWN!!!!!"

I am on Ubuntu Rockchip (desktop) with the latest kernel 5.10.125. Do you happen to know, if that "changed kernel API" problem occurs in that case?

Joshua-Riek, 2023-12-19:
There is a new Linux 6.1 BSP that should be released sometime this month. 

I can try to fix this during the rebase process. I should just need to drop the  8852be driver commit and use the proper  rtw89 dkms.

bagong, 2023-12-19:
Okay, that sounds great. I can live with this for the time being as it's basically fixed by shutting down one of the devices. Thanks for your attention!

closed 2023-12-19 by bagong