Joshua-Riek/ubuntu-rockchip #521: Unable to wake after suspending

milkmuffin, 2023-12-19:
Image installed: `ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5-plus.img.xz`

Hardware: [Orange Pi 5 Plus (32GB)](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-plus-32GB.html)

When I suspend, the screen goes black but then I am unable to wake the computer. (I would expect it should wake upon hitting the power button).

I have to do a hard reset (cycle the power) & reboot to get it back.

milkmuffin, 2023-12-19:
It seems to be working now. I'm not able to reproduce the original problem. Thank you.

milkmuffin, 2023-12-19:
It appears to suspend correctly on the very first boot (right after installation).

On subsequent boots, it no longer suspends correctly.

Joshua-Riek, 2023-12-19:
I know there are new DDR and BL31 blobs (bootloader firmware) that will fix some issues with suspended systems. But I can't upgrade these, or users with other distros such as Armbian or another os could crash and have stability issues.

But this could just be one small thing required. It could also be the kernel driver. However, I will not spend time looking into this as a new BSP with Linux 6.1 should be released sometime this month. I'm really only working on critical issues until the new BSP release comes out.

bagong, 2023-12-22:
I have the same opi5+ model and use a bluetooth connected keyboard to wake from suspend and it seems to work fine. Is it the expected behaviour to wake from suspend with the power button? When I press the power button while the computer is running normally, I get the dialog that offers shutdown.

closed 2025-02-17 by github-actions[bot]