Joshua-Riek/ubuntu-rockchip #523: OPI 5 plus freezes

Davidshu8611, 2023-12-19:
Ubuntu 22.04 3 LTS for Orange PI 5 plus  freezes from time to time for no reason...  log txt   as below:

Dec  5 15:50:01 dawei-ai25007 CRON[1966]: (ubuntu) CMD (/etc/zabbix/parameter/software/server_status)
Dec  5 15:50:02 dawei-ai25007 CRON[1964]: (CRON) info (No MTA installed, discarding output)
Dec  5 15:50:56 dawei-ai25007 mpp[1304]: rc_model_v2: invalid fix 0 qp found set default qp 26
Dec  5 15:50:58 dawei-ai25007 mpp[1304]: message repeated 5 times: [ rc_model_v2: invalid fix 0 qp found set default qp 26]
Dec  5 15:51:01 dawei-ai25007 CRON[1978]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec  5 15:51:01 dawei-ai25007 CRON[1977]: (CRON) info (No MTA installed, discarding output)
Dec  5 15:54:01 dawei-ai25007 CRON[2010]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec  5 15:54:01 dawei-ai25007 CRON[2009]: (CRON) info (No MTA installed, discarding output)
Dec  5 15:57:01 dawei-ai25007 CRON[2027]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec  5 15:57:01 dawei-ai25007 CRON[2026]: (CRON) info (No MTA installed, discarding output)
Dec  5 15:59:01 dawei-ai25007 CRON[2047]: (ubuntu) CMD (/ssd/manage/ffmpeg_video/save_data.py normal >> /ssd/manage/ffmpeg_video/save.log 2>&1)
Dec  5 15:59:29 dawei-ai25007 mpp[1304]: rc_model_v2: invalid fix 0 qp found set default qp 26
Dec  5 15:59:38 dawei-ai25007 mpp[1304]: message repeated 51 times: [ rc_model_v2: invalid fix 0 qp found set default qp 26]
Dec  5 16:00:01 dawei-ai25007 CRON[2109]: (ubuntu) CMD (/etc/zabbix/parameter/software/server_status)
Dec  5 16:00:01 dawei-ai25007 CRON[2110]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec  5 16:00:01 dawei-ai25007 CRON[2107]: (CRON) info (No MTA installed, discarding output)
Dec  5 16:00:01 dawei-ai25007 CRON[2108]: (CRON) info (No MTA installed, discarding output)
Dec  5 16:00:02 dawei-ai25007 kernel: [ 2246.033683] perf: interrupt took too long (3134 > 3131), lowering kernel.perf_event_max_sample_rate to 63600
Dec  5 16:00:21 dawei-ai25007 mpp[1304]: rc_model_v2: invalid fix 0 qp found set default qp 26
Dec  5 16:01:29 dawei-ai25007 mpp[1304]: message repeated 76 times: [ rc_model_v2: invalid fix 0 qp found set default qp 26]
Dec  5 16:03:01 dawei-ai25007 CRON[2190]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec  5 16:03:01 dawei-ai25007 CRON[2189]: (CRON) info (No MTA installed, discarding output)
Dec  5 16:04:52 dawei-ai25007 systemd[1]: Started Session 27 of User ubuntu.

Joshua-Riek, 2023-12-19:
Looks like your running a script? But I don't see any crash. Please upload output of journalctl in a text file.

Davidshu8611, 2023-12-19:
[25007-syslog.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13716416/25007-syslog.txt)

Davidshu8611, 2023-12-19:
[25001-syslog.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13716422/25001-syslog.txt)

Davidshu8611, 2023-12-19:
The status of the motherboard is red, and the normal operation is red and blue alternately, and then the dual network port indicator blinks, but there is no data output, so it is judged to be a device crash

Joshua-Riek, 2023-12-19:
I don't see any obvious problems besides what I assume are some of your scripts running? If your system is crashing it looks like the below script or some other process you created is causing the problem.

/ssd/eagle_skip/scripts/remote/remote_check

Davidshu8611, 2023-12-20:
But  I have  another 4 pcs  OPI 5plus ,they run the same scripts, but didn't  crashed until today , only the two pcs crashed, so ----

Davidshu8611, 2023-12-20:
25007  Device froze at Dec 2 12:20:02 
25001 Device froze at Dec 3 16:09:01 After these times, there were no system logs printed , The logs from Dec 5 are after a manual power cycle.

Joshua-Riek, 2023-12-20:
You have 4 Opi5+ that run the same software fine, but one will freeze? Are they all the same RAM size?

Davidshu8611, 2023-12-20:
total  6pcs OPi5+  ,2 pcs crashed, yes ,they are the same RAM size ,run same software.

Joshua-Riek, 2023-12-20:
I dont have enough information to tell what is going on. The only thing strange i see is that a SPI transfer keeps happening and timing out. I would assume this is done by your custom software.

```
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501253] rockchip-spi feb20000.spi: RK SPI transfer timed out
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501321] rk806 spi2.0: SPI transfer failed: -110
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501344] rockchip-spi feb20000.spi: state=0
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501360] rockchip-spi feb20000.spi: tx_left=0
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501377] rockchip-spi feb20000.spi: rx_left=3
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501398] regs 00000000: 00002c01 00000002 00000000 00000000
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501418] regs 00000010: 000000c6 00000000 00000002 00000000
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501437] regs 00000020: 00000000 0000004c 00000000 00000000
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501457] regs 00000030: 00000000 00000001 00000001 00000000
Dec  3 13:20:12 dawei-ai25001 kernel: [47917.501475] regs 00000040: 0000001f 00000000 00110002
```

Joshua-Riek, 2023-12-20:
Please try to isolate what is going on, as I see no obvious crashing. I also have no idea what your use case is or what is being preformed on your systems. If more information is not provided I will close this issue, I don't have time for this.

Davidshu8611, 2023-12-25:
[syslog.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13765844/syslog.txt)
Please help analyze the log file, and it has crashed again. Our use scenario is that this device acts as a edge computing server, and the terminal device transmits video streams to do AI intelligent analysis, such as rope skipping, long jump and other sports scenes

Davidshu8611, 2023-12-25:
We use IP camera and HDMI interfaces to obtain video streams and display results, without using the SPI interface of OPI 5+. Therefore, can we add protection at the system layer so that if SPI transmission occurs, the process can be terminated directly,

Joshua-Riek, 2023-12-25:
I still do not see a crash. Maybe MPP or ffmpeg is leaking memory? Are you able to run some testing without using hardware acceleration? 

Also, is the below where your system crashed? 
```
Dec 24 07:50:01 dawei-ai25996 CRON[177052]: (ubuntu) CMD (/etc/zabbix/parameter/software/server_status)
Dec 24 07:50:01 dawei-ai25996 CRON[177051]: (CRON) info (No MTA installed, discarding output)
Dec 24 07:51:01 dawei-ai25996 CRON[177165]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec 24 07:51:01 dawei-ai25996 CRON[177164]: (CRON) info (No MTA installed, discarding output)
Dec 24 07:51:42 dawei-ai25996 mpp[71363]: H265_PARSER_REF: cur_poc 18 Could not find ref with POC 17
Dec 25 09:29:28 dawei-ai25996 fake-hwclock[448]: Current system time: 2023-12-25 01:29:15
Dec 25 09:29:28 dawei-ai25996 fake-hwclock[448]: fake-hwclock saved clock information is in the past: 2023-12-23 23:17:01
Dec 25 09:29:28 dawei-ai25996 fake-hwclock[448]: To set system time to this saved clock anyway, use "force"
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Mounted Kernel Debug File System.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Mounted Kernel Trace File System.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Restore / save the current clock.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Set the console keyboard layout.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Create List of Static Device Nodes.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: modprobe@configfs.service: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module configfs.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: modprobe@drm.service: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module drm.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: modprobe@efi_pstore.service: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module efi_pstore.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: modprobe@fuse.service: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module fuse.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Modules.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Remount Root and Kernel File Systems.
Dec 25 09:29:28 dawei-ai25996 kernel: [    1.775583] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Activating swap /swapfile...
```

Davidshu8611, 2023-12-25:
yes

---- Replied Message ----
| From | Joshua ***@***.***> |
| Date | 12/26/2023 00:43 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Joshua-Riek/ubuntu-rockchip] OPI 5 plus freezes (Issue #523) |

I still do not see a crash. Maybe MPP or ffmpeg is leaking memory? Are you able to run some testing without using hardware acceleration?

Also, is the below where your system crashed?

Dec 24 07:50:01 dawei-ai25996 CRON[177052]: (ubuntu) CMD (/etc/zabbix/parameter/software/server_status)
Dec 24 07:50:01 dawei-ai25996 CRON[177051]: (CRON) info (No MTA installed, discarding output)
Dec 24 07:51:01 dawei-ai25996 CRON[177165]: (root) CMD (/ssd/eagle_skip/scripts/remote/remote_check)
Dec 24 07:51:01 dawei-ai25996 CRON[177164]: (CRON) info (No MTA installed, discarding output)
Dec 24 07:51:42 dawei-ai25996 mpp[71363]: H265_PARSER_REF: cur_poc 18 Could not find ref with POC 17
Dec 25 09:29:28 dawei-ai25996 fake-hwclock[448]: Current system time: 2023-12-25 01:29:15
Dec 25 09:29:28 dawei-ai25996 fake-hwclock[448]: fake-hwclock saved clock information is in the past: 2023-12-23 23:17:01
Dec 25 09:29:28 dawei-ai25996 fake-hwclock[448]: To set system time to this saved clock anyway, use "force"
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Mounted Kernel Debug File System.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Mounted Kernel Trace File System.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Restore / save the current clock.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Set the console keyboard layout.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Create List of Static Device Nodes.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: ***@***.***: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module configfs.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: ***@***.***: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module drm.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: ***@***.***_pstore.service: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module efi_pstore.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: ***@***.***: Deactivated successfully.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Module fuse.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Load Kernel Modules.
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Finished Remount Root and Kernel File Systems.
Dec 25 09:29:28 dawei-ai25996 kernel: [    1.775583] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Dec 25 09:29:28 dawei-ai25996 systemd[1]: Activating swap /swapfile...

ironiris, 2023-12-26:
Try changing the storage device just in case.(Micro SD card or NVMe)

Davidshu8611, 2023-12-27:
> I still do not see a crash. Maybe MPP or ffmpeg is leaking memory? Are you able to run some testing without using hardware acceleration?
>
> Also, is the below where your system crashed?
>
> ```
> Dec 24 [...]

Please tell me how to turn on the hardware watchdog function

Joshua-Riek, 2024-01-12:
Closing as stale and unable to reproduce.

closed 2024-01-12 by Joshua-Riek

mariusseitan, 2024-03-04:
I also have this problem, and I saw that many people had it with Orange Pi 5 Plus in the past, but nothing is solved. My SBC freezes from time to time, especially if I'm using remote desktop to control the SBC's desktop from my Windows PC. But this is happening only if the OS is flashed on nvme. I've noticed that I never had this problem if I'm using the same OS on eMMC. Could it be because of the poor nvme disk quality? I don't think it is because of the power supply. I use one that have 5V 4A.

mariusseitan, 2024-03-04:
Also, after the crash, i performed a hard reboot, and when i run "journalctl > sys.log", i get "Journal file /var/log/journal/5d491fe56ba79a649646b63f65712cb8/system@0006128470a97410-15d30698eabf9776.journal~ is truncated, ignoring file."

ironiris, 2024-03-04:
> I also have this problem, and I saw that many people had it with Orange Pi 5 Plus in the past, but nothing is solved. My SBC freezes from time to time, especially if I'm using remote desktop to [...]

As I mentioned above, try changing the storage device. I've seen several cases where the device goes down for unknown reasons due to a storage device issue. This is a problem not only with the Orange Pi 5 but also seen with x86 PCs.

SimplyCorbett, 2024-03-21:
Are the people having this issue on here using the built in 1GB swap file on the SD card? Just curious.