Joshua-Riek/ubuntu-rockchip #527: RockPI5 NPU .so missing

anchor76, 2023-12-24:
hello,

i'm trying to perform some opengl rendering, but getting this runtime error:

libEGL warning: MESA-LOADER: failed to open rknpu: /usr/lib/dri/rknpu_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

glxinfo brief oputput:
"
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Panfrost (0xffffffff)
    Device: Mali-G610 (Panfrost) (0xffffffff)
    Version: 23.0.0
    Accelerated: yes
    Video memory: 3919MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.3
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Panfrost
OpenGL renderer string: Mali-G610 (Panfrost)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 23.0.0-devel
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
"

OpenGL version string: 3.3 (Compatibility Profile) Mesa 23.0.0-devel
OpenGL shading language version string: 3.30
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.0.0-devel
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10

board: rockpi5b
ubuntu: Linux ubuntu 5.10.160-rockchip #26 SMP Thu Dec 21 14:15:31 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

how to install the missing NPU driver?

Joshua-Riek, 2023-12-24:
The NPU is not at all related to Mesa or the GPU.

I'm confused about how you stumbled across this, please provide the exact steps to reproduce the issue.

anchor76, 2023-12-24:
hello, thanks for your response.

steps to reproduce the issue:

1 - download and write this image to rockpi5b emmc: ubuntu-22.04.3-preinstalled-server-arm64-rock-5b.img
2 - apt update, apt upgrade
3 - apt install -y xorg openbox mesa-utils build-essential libx11-dev freeglut3-dev zlib1g-dev
4 - compile this headless example: https://github.com/erwincoumans/egl_example
5 - nohup startx &
6 - export DISPLAY=:0
7 - run the binary, observe the output.

'libEGL warning: MESA-LOADER: failed to open rknpu: /usr/lib/dri/rknpu_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)'

(please note, i'm on beginner level with linux.)

Joshua-Riek, 2023-12-24:
Please know the mesa version used by this system is a hacked up mess, so be careful.

I'm not very familiar with rknpu, but let me checkout the Rockchip SDK again, it will take me maybe 4 hours since its so large. But once downloaded I can look to see if there are any libs with the name of rknpu_dri.so.

Joshua-Riek, 2023-12-24:
I do see this repo, it may contain what you are looking for https://github.com/rockchip-linux/rknpu2

anchor76, 2023-12-24:
thanks i'll check it. meanwhile i tried the radxa official ubuntu image with the same result. or worse, because glxinfo see only LLVM pipe ..., no gpu accelerated rendering.

Joshua-Riek, 2023-12-24:
That sounds about right, the software/drivers for this platform are very bad. I'm not sure RK3588 can use rknpu_dri.so

anchor76, 2023-12-24:
meanwhile i tried radxa official debian image, same result. and only llvmpipe. what you @Joshua-Riek released, that ubuntu can use the mail GPU. seems the most feature complete image for this board.

my goal is: headless opengl rendering on a linux server. (ubuntu is preferred.) can you please recommend a board with much better software/driver support? i believe raspberry pi line has very good support, but the hardware is weak... :)

Joshua-Riek, 2023-12-24:
I don't know of any cheap boards for that task with better software and driver support. The other boards I know of are easily $1,000 and up such as the Honeycomb LX2.

anchor76, 2023-12-25:
@Joshua-Riek
Radxa clarified the situation here: https://github.com/radxa-build/rock-5b/issues/9
If i understood correctly, currently only the desktop builds are configured properly for OpenGL(ES) rendering, and configuring the upstream build is complicated. I'll use my passive cooled Athlon 3000G board until i find a better solution.

Thank you, i'm closing this issue.

closed 2023-12-25 by anchor76

RadxaYuntian, 2023-12-25:
I see the full log now via the issue x-ref. I thought the kernel used here was a upstream kernel with panthor so I told @anchor76 that NPU is not supported in upstream kernel, but this is a vendor kernel, so NPU should work here.

I have no idea what `rknpu_dri.so` is, but we don't have it in our system and [RKNN2 still works](https://docs.radxa.com/en/radxa-os/development/rknn-toolkit-lite2). This is only preinstalled in recent desktop systems which might be why you didn't get it working initially, but you can do system upgrade and the correct package will be installed for you.

You can download [our packaged RKNN2 debs](https://github.com/radxa-pkg/rknn2/releases/tag/1.6.0-1) and install on Joshua's Ubuntu and see if they will work.

anchor76, 2023-12-27:
@RadxaYuntian - thanks for your response. i tried to install the mentioned RKNN2 .deb package, but didn't work. some opengl component is still looking for rknpu_dri.so lib.

content of RKNN2 .deb package for RK3588 chip:
[image]

RadxaYuntian, 2023-12-28:
Like I said I have no idea what `rknpu_dri.so` is, so it is understandable that you won't find that file in our package. That package should allow you to run NPU related stuff though, which is not the same as OpenGL.

anchor76, 2024-03-02:
Finally i found the solution: on RK3588 EGL see mutiple render devices, and the first is the NPU for some reason. I just need to use device https://github.com/radxa-build/rock-5b/issues/1 instead of device #0. Problem solved ;)

MyVizDrake, 2024-07-17:
@anchor76 I am seeing the same error when running mpv (#931). Not sure my issue is the same as what you were experiencing but it's the same error.  How did you force device number 1 instead of device number 0?  Complete embedded linux noob here :-)

anchor76, 2024-07-17:
@MyVizDrake i'm using OpenGL from C++ code, so my solution was a code solution. i'm also not a linux expert, so i can just assume: it's maybe somehow possible to disable the device #0 (NPU), and then the apps will use the device #1 (GPU) as default. 
a linux guru can tell it.