Joshua-Riek/ubuntu-rockchip #529: Problem when running build.sh on Vanilla Ubuntu 22.04 image for Orange Pi 5 Plus(camera_egine_rkaiq)

aramlow, 2023-12-26:
I have tried several times on distilled it down to a plain Ubuntu 22.04 server install and following the commands on

It fails with the following error message

```
dpkg-deb: error: '/tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb' is not a Debian format archive
dpkg: error processing archive /tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb
Error: in ./scripts/build-rootfs.sh on line 260
Error: in ./build.sh on line 167
```
It does get past it if I comment out the relevant lines in the `build-rootfs.sh` file, since I don't need the camera for my use-case.

hakotani-o, 2023-12-26:
Please download packages/rkaiq/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb again.

Joshua-Riek, 2023-12-26:
You did not checkout the git lfs artifacts.
```
git lfs fetch && git lfs checkout
```

closed 2023-12-30 by Joshua-Riek