Joshua-Riek/ubuntu-rockchip #530: orangepi-config: command not found

3diamond, 2023-12-27:
HI
I hate to do this but I am stumped.
I am trying to access orangepi-config so I can use the orange pi 5 plus to work with the official lcd touchscreen. does this version of ubuntu v1.31 have orangepi-config? if so what is the problem?

Also many thanks for Joshua and unsung heroes.

Ferenc-, 2023-12-27:
This has been asked and answered  before: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/387

Joshua-Riek, 2023-12-27:
The orangepi-config tool is from Armbian but with a diffrent name. Since this is not Armbian it does not and will not exist.

closed 2023-12-27 by Joshua-Riek

3diamond, 2023-12-27:
Ok thanks for your quick reply and for letting me know. Is there a way to activate the lcd port?

Joshua-Riek, 2023-12-27:
You can enable with the device tree overlay. 

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

3diamond, 2023-12-27:
Thank very much for taking the time in your busy schedule to help. I'll check it out.