Joshua-Riek/ubuntu-rockchip #532: Can you provide assistance related to the kernel below?

124ffee, 2023-12-28:
My device is Orange Pi 5 Plus, and I am using your Ubuntu system. I am running a Redroid Android Docker container on the system. Now, I want to grant it root permissions with KernelSU, and I need the corresponding kernel. Can you provide me with some relevant ideas and assistance?

Joshua-Riek, 2023-12-31:
I have no idea what this is or what you want.

closed 2023-12-31 by Joshua-Riek

Ferenc-, 2024-01-01:
My interpretation of what @124ffee might have wanted here:

The [KernelSU](https://github.com/tiann/KernelSU) project is a root management solution for Android.
This `KernelSU` mentions supports for `"container-based Android"`, hence the speculation, that it could theoretically work with [remote-android](https://github.com/remote-android) running on a [Joshua-Riek/ubuntu-rockchip](https://github.com/Joshua-Riek/ubuntu-rockchip) system.

In order to make use of `KernelSU` with a non-whitelisted kernels, one has to build a kernel with the `KernelSU` kernel driver, and one would need the the matching kernel (sources) for that.

If all that speculation is true, then the kernel sources are here:
https://github.com/Joshua-Riek/linux-rockchip
But there is no further assistance on building the `KernelSU` drivers.