Joshua-Riek/ubuntu-rockchip #533: Ubuntu Desktop V1.32 on Orange Pi 5 Plus cannot boot from SDCard

levinhtin, 2023-12-29:
I have flashed many times with Ubuntu Desktop for Orange Pi 5 plus v1.32 but cannot boot from SDCard.
Then, I was tried with v1.30, v.131. They work normally.

bagong, 2023-12-29:
Can confirm. Both the default desktop-build as well as the mainline server build of the 1.32 release have empty /boot/firmware folders.

Joshua-Riek, 2023-12-29:
Oh, I found the issue, it was a typo. I had the Opi5+ trying to load a rk3588s device tree. I'm running a new release build now and should be fixed in about 1 hour.

Joshua-Riek, 2023-12-29:
Fixed, please look at the updated release.

closed 2023-12-29 by Joshua-Riek

bagong, 2023-12-29:
Confirmed fixed. You might want to trigger the mainline build as well, it suffers from the same calamity ;-)