Joshua-Riek/ubuntu-rockchip #536: Can you provide an Ubuntu image for the nvme version of rock 5b? Thank you very much!

songsong-sunny, 2023-12-30:
Can you provide an Ubuntu image for the nvme version of rock 5b? Thank you very much!

Joshua-Riek, 2023-12-31:
NVMe boot works perfectly on the ROCK 5B. Please follow the below steps:

1. Boot from an sdcard
2. Update bootloader on SPI Flash
```
sudo u-boot-install-mtd /dev/mtdblock0
```
3. Install ubuntu on the NVMe 
```
sudo ubuntu-rockchip-install /dev/nvme0n1
```

KeiroD, 2024-01-05:
I was having similar issues because I was expecting to just flash this onto the NVMe and be up and running because I'd already had my SPI set up to allow for NVMe boot with other OSes. That's what I get for assuming I can just skip the instructions to use this on SD card first as noted in the read-me. :P

Perhaps this should be noted in the read-me?

However, following the instructions here allowed me to get Ubuntu onto the NVMe as expected. Works exactly like you'd configure from the SD card.

develperbayman, 2024-01-08:
i know not everyone has a adapter but i just used a nvme to usb and used etcher was the easiest

closed 2024-01-17 by Joshua-Riek

rainingcrypto, 2024-04-02:
> NVMe boot works perfectly on the ROCK 5B. Please follow the below steps:
> 
> 1. Boot from an sdcard
> 2. Update bootloader on SPI Flash
> 
> ```
> sudo u-boot-install-mtd /dev/mtdblock0
> ```
> 
> 3. Install ubuntu on the NVMe
> 
> ```
> sudo ubuntu-rockchip-install /dev/nvme0n1
> ```

hello "our lord and savior" Joshua... these two commands, along with the instruction to install first to a micro SD card, and then boot from that - literally saved my life. 
Would you be so kind to add a section to the main Read Me, titled "Booting from NVME" and put this in there? It would save idiots like the trouble of digging through the issues. 

Anyway - because of you I'm a happy man as my Rock 5B 32GB ram is running like a champ. Cheers!