Joshua-Riek/ubuntu-rockchip #538: APP armor - DEAD

seler69, 2023-12-30:
Hello everyone!

im trying to install Homeassistant Supervised on Joshuas ubuntu RK3588 edition.
i aleready made it but ...I have one last thing left to be completely happy this year!

APPARMOR - Active!

im trying on desktop and server edition and i cant get it done.
Ubuntu should have apparmor already instaled?
Re-install apparmor and profiles didnt work!

i was trying to add some lines in Bootenv
Run armbian-config (orangepi-config etc)

    System → Bootenv. Add the following lines:

    extraargs=apparmor=1 security=apparmor
    systemd.unified_cgroup_hierarchy=0

BUT I CANT FIND SOMETING SIMILAR TO "armbian-config" in ubuntu
ubuntu-config dosen't exist

root@ubuntu:/home/ubuntu# systemctl status apparmor.service
○ apparmor.service - Load AppArmor profiles
     Loaded: loaded (/lib/systemd/system/apparmor.service; enabled; vendor pres>
     Active: inactive (dead)
  Condition: start condition failed at Fri 2023-12-29 16:10:21 UTC; 1 day 2h ago
       Docs: man:apparmor(7)

Help please!

Joshua-Riek, 2023-12-30:
You can add the kernel command line params to the file /boot/firmware/ubuntuEnv.txt. Please see the wiki page below.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Kernel-Command-Line

seler69, 2023-12-30:
>
     Active: active (exited) since Sat 2023-12-30 20:23:05 CET; 1min 18s ago
       Docs: man:apparmor(7)
             https://gitlab.com/apparmor/apparmor/wikis/home/
    Process: 640 ExecStart=/lib/apparmor/apparmor.systemd reload (code=exited, >
   Main PID: 640 (code=exited, status=0/SUCCESS)

Thank You! im angry the answer was on Your page! sorry to bother You!

seler69, 2023-12-30:
Active: active (exited) since Sat 2023-12-30 20:23:05 CET; 1min 18s ago
   Docs: man:apparmor(7)
         https://gitlab.com/apparmor/apparmor/wikis/home/
Process: 640 ExecStart=/lib/apparmor/apparmor.systemd reload (code=exited, >

Main PID: 640 (code=exited, status=0/SUCCESS)

Thank You! im angry the answer was on Your page! sorry to bother You!

closed 2023-12-30 by seler69

flyzet-prog, 2024-01-24:
> Active: active (exited) since Sat 2023-12-30 20:23:05 CET; 1min 18s ago Docs: man:apparmor(7) https://gitlab.com/apparmor/apparmor/wikis/home/ Process: 640 ExecStart=/lib/apparmor/apparmor.systemd [...]

Hi seler69. My device OPi5plus 16GB. How did you resolved this issue with apparmor. 
I added to /boot/firmware/ubuntuEnv.txt exactly this code `extraargs=apparmor=1 security=apparmor
systemd.unified_cgroup_hierarchy=0`
But still shows "inactive(dead)"
`○ apparmor.service - Load AppArmor profiles
     Loaded: loaded (/lib/systemd/system/apparmor.service; enabled; vendor pres>
     Active: inactive (dead)
  Condition: start condition failed at Wed 2024-01-24 13:40:16 EET; 46s ago
             └─ ConditionSecurity=apparmor was not met
       Docs: man:apparmor(7)
             https://gitlab.com/apparmor/apparmor/wikis/home/
`

Would be appreciated for any help.

seler69, 2024-01-24:
GNU nano 6.2                                                                                           /boot/firmware/ubuntuEnv.txt
bootargs=root=UUID=d6250cd5-a4ca-49e2-b0b8-0ba7bb333163 rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 extraargs=apparmor=1  security=apparmor systemd.unified_cgroup_hierarchy=0 quiet splash plymouth.ignore-serial-consoles
fdtfile=rk3588s-orangepi-5.dtb
overlay_prefix=rk3588
overlays=

thats mine ENV file
1. check spelling
2. put 'extraargs=apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=0'' in 1st line like me.
a get similar errors when i put extraags in different places in ENV file. just put it in first line.
4. save file
5. reboot!!
remeber to reboot!!!

flyzet-prog, 2024-01-24:
> GNU nano 6.2 /boot/firmware/ubuntuEnv.txt bootargs=root=UUID=d6250cd5-a4ca-49e2-b0b8-0ba7bb333163 rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 extraargs=apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=0 quiet splash plymouth.ignore-serial-consoles fdtfile=rk3588s-orangepi-5.dtb overlay_prefix=rk3588 overlays=
> 
> thats mine ENV file
> 
> 1. check spelling
> 2. put 'extraargs=apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=0'' in 1st line like me.
>    a get similar errors when i put extraags in different places in ENV file. just put it in first line.
> 3. save file
> 4. reboot!!
>    remeber to reboot!!!

Finally it's works. Thank you for your suggestions.