Joshua-Riek/ubuntu-rockchip #539: Cannot access PWM on two pins on Nanopc T6

Xeue, 2023-12-30:
I've been trying to use PWM on the T6 and can't seem to access pin 8 (GPIO0_C5 - 21 - PWM4_M0) or pin 32 (GPIO0_C6 - 22 - PWM5_M1)
I can see the rest of the pins.
In sys/class/pwm ther is only a pwmchip0. On the other boards there are also pwmchip1 and pwmchip2
Looking at device overlays, there are no dtbo files for the relevant pins. However some of the pins that are working don't have any dtbo files either...

For the orange pi 5 +, in the wiringNP library it has the folllowing for the addresses for the controllers:
#define RK3588_PWM0_BASE                        0xfd8b0000U
#define RK3588_PWM1_BASE                        0xfe8d0000U
#define RK3588_PWM2_BASE                        0xfebe0000U
#define RK3588_PWM3_BASE                        0xfebf0000U

For the T6 I think it should be "RK3588_PWM1_BASE                        0xfe8d0000U" for PWM1, but that doesn't help.

I'm new to all of this so don't really understand what I'm looking at, but I feels like something is missing and some hardware isn't added somwhere...

nyanmisaka, 2023-12-31:
GPIO support was added by upstream friendlyelec in https://github.com/friendlyarm/kernel-rockchip/commit/69858afd5077a51449ccf6b3c736988221687547 but it [broke](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/398) the T6 in ubuntu-rockchip. So it has been reverted in https://github.com/armbian/linux-rockchip/commit/80b56be390b6321fe9690544419aabf766b13c11

Joshua-Riek, 2024-01-02:
I can take a look at this tomorrow and double-check with the schematics, it's likely there is a misconfigured pinctrl. Or at the very least I can enable the PWM nodes in the device tree.

Xeue, 2024-01-02:
Enabling the PWM nodes will certainly be a start, the rest I can do myself (maybe!)

Joshua-Riek, 2024-01-03:
This should enable pwm 4 and 5, I can't test on the hardware right now but looks correct according to the schematic. Over the weekend ill take a look again at the upstream patch from FriendlyELEC and see if i can figure out why it was causing a crash in the first place.

```diff
diff --git a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
index 1624b28f0529..5795f6f941e8 100644
--- a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
@@ -585,6 +585,16 @@ &pwm2 {
        /* connected with MIPI-DSI0 */
 };
 
+&pwm4 {
+       pinctrl-0 = <&pwm4m0_pins>;
+       status = "okay";
+};
+
+&pwm5 {
+       pinctrl-0 = <&pwm5m1_pins>;
+       status = "okay";
+};
+
 &pwm11 {
        pinctrl-0 = <&pwm11m3_pins>;
        /* connected with MIPI-DSI1 */
```

closed 2025-02-17 by github-actions[bot]