Joshua-Riek/ubuntu-rockchip #541: Upgrade kernel without reimagining the entire OS

denis-itskovich, 2024-01-01:
Is it possible to upgrade kernel without replacing the entire OS image? If it is possible, I think it worths do add instructions into `README.md`

Ferenc-, 2024-01-01:
Among the [Highlights](https://github.com/Joshua-Riek/ubuntu-rockchip?tab=readme-ov-file#highlights) one can read
`Receive kernel, firmware, and bootloader updates through apt`

The kernel is available in this pre-configured PPA `https://ppa.launchpadcontent.net/jjriek/rockchip/ubuntu`

closed 2024-01-01 by denis-itskovich

rawsome1, 2024-01-08:
> Is it possible to upgrade kernel without replacing the entire OS image? If it is possible, I think it worths do add instructions into `README.md`

Hello, did you get a reply to your question and if you did, would share it with me?

Joshua-Riek, 2024-01-08:
Sorry I forgot to get back to you. To upgrade the kernel, simply upgrade your packages as per normal for ubuntu. The exact commands are below.

```
sudo apt update 
sudo apt upgrade
```