Joshua-Riek/ubuntu-rockchip #544: Ubuntu server user password is wrong and IP cannot be obtained

DHDAXCW, 2024-01-02:
Hello, I came to report a server problem and found that the account and password entered when testing the rock-5a Ubuntu Server were wrong. After I changed the account and password to enter, the IP cannot access the Internet. Ifconfig sees only

The user and password entered below are wrong

```
[    8.758938] systemd[1]: modprobe@efi_pstore.service: Deactivated successfully.
[    8.760653] systemd[1]: Finished Load Kernel Module efi_pstore.
[    8.763540] systemd[1]: modprobe@fuse.service: Deactivated successfully.
[    8.765219] systemd[1]: Finished Load Kernel Module fuse.
[    8.768406] systemd[1]: Finished Load Kernel Modules.
[    8.771178] systemd[1]: Started Journal Service.
[    8.793229] device-mapper: ioctl: 4.44.0-ioctl (2021-02-01) initialised: dm-devel@redhat.com
[    8.800985] systemd-journald[433]: Received client request to flush runtime journal.
[    8.812528] Adding 2097148k swap on /swapfile.  Priority:-2 extents:6 across:2965500k SS

Ubuntu 22.04.3 LTS ubuntu ttyFIQ0

ubuntu login: ubuntu
Password:

Login incorrect
ubuntu login: ubuntu
Password:

Login incorrect
ubuntu login:
Login timed out after 60 seconds.

Ubuntu 22.04.3 LTS ubuntu ttyFIQ0

ubuntu login:
```

IP cannot obtain IP, execute `sudo dhclient` to solve the problem
```
ubuntu@ubuntu:~$ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 342  bytes 24724 (24.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 342  bytes 24724 (24.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ubuntu@ubuntu:~$ sudo dhclient
[sudo] password for ubuntu:
ubuntu@ubuntu:~$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.173  netmask 255.255.255.0  broadcast 192.168.3.255
        inet6 fe80::4029:3bff:fe94:14d1  prefixlen 64  scopeid 0x20<link>
        ether 42:29:3b:94:14:d1  txqueuelen 1000  (Ethernet)
        RX packets 27  bytes 2307 (2.3 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 9  bytes 1270 (1.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 89

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 842  bytes 60276 (60.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 842  bytes 60276 (60.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlP4p65s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 78:8a:86:34:22:d6  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 7a:8a:86:34:22:d6  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan2: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 72:8a:86:34:22:d6  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ubuntu@ubuntu:~$
```
I only tested rock-5a and had problems. I don’t know if other machines have problems like mine?

Thanks

DHDAXCW, 2024-01-02:
I tested the desktop version with no problems

erd-yunmu, 2024-01-02:
Please try this method to fix your server unable to obtain IP after booting. https://github.com/erd-yunmu/ubuntu-rockchip/commit/631797ba2b97be9f7c1e135dd1ea949e8388b455
As for Ubuntu password errors, I set my own username and password. Did this help you? https://github.com/erd-yunmu/ubuntu-rockchip/commit/042cf22cb1c76848f68589acf5342fa488bdefcd

^v^

DHDAXCW, 2024-01-03:
@erd-yunmu Thank you very much, these two methods have solved my problem, thank you!

DHDAXCW, 2024-01-03:
@Joshua-Riek I found that the server enters the account and password normally on HDMI. But when inputting on the ttl serial port, it prompts that the password is incorrect.

closed 2024-01-16 by DHDAXCW

Joshua-Riek, 2024-01-16:
Ubuntu server needs about 30-60 seconds before it creates the user account and password. Sorry about the delay for this, been busy with other tasks and the holidays.

netPokey, 2024-04-23:
> @Joshua-Riek发现服务器在HDMI上输入账号密码正常。但在ttl串口输入时，提示密码错误。

ttl 密码错误如何解决呢