Joshua-Riek/ubuntu-rockchip #548: No HDMI sound out on OrangePi 5B

apsinghnandha, 2024-01-03:
Hi,
I can see audio device playing but no sound over HDMI. I tried last two versions of Gnome. Then i came across this comment [link](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/429#issuecomment-1793020616) and i was able to solve the issue by installing pipewire for audio. But it would be great if the issue isn't there at first place

bagong, 2024-01-03:
Pipewire is the future anyways, but: the problem must lie on your side, as sound via hdmi works. It worked in all my installs and the fora would be full of complaints if it didn't. I think you either didn't choose the right audio out device, or you have some deeper misconfiguration.

Joshua-Riek, 2024-01-03:
Audio works perfectly fine over HDMI for me this problem must be some sort of config error on your end.

closed 2024-01-03 by Joshua-Riek