Joshua-Riek/ubuntu-rockchip #549: Erratic ethernet operation on orange pi 5B

PteronuraBrasiliensis, 2024-01-03:
Hello, i am using the release 1.32 of the desktop version for the orange pi 5B. I am trying to use the ethernet port but it doesn't work properly. When trying to connect to the internet, no IP can be assigned via DHCP. I also tried connecting a vlp-16 lidar sensor which only works by assigning the IP manually. After setting the IP address, the system is able to communicate with the sensor for about 10 seconds before disconnecting. I have to manually unplug/plug the cable for the connection to work again.

I am not a linux expert, but i saw these error messages when booting and it seems to be related to this problem:
[image]

I know the dhcp part is a known issue but i wanted to add that it doesn't work on manual IP either. Thank you in advance !

Joshua-Riek, 2024-01-03:
Sadly those dmesg logs mean little since the kernel is a hack in the first place. I will grab my OPi5B and test on my end.

Joshua-Riek, 2024-01-03:
I just tested with the latest build and ethernet is working perfectly fine for me. Maybe your network has some sort of issue.

closed 2024-01-03 by Joshua-Riek