Joshua-Riek/ubuntu-rockchip #561: NanoPC-T6 no more audio via HDMI

almaster76, 2024-01-11:
After normal upgrade with apt, no more audio via HDMI is played with my NanoPC-T6 (server version). This happens in KODI and RetroArch. The soundcard seems to be correctly recognized (aplay -l), but no sound is played via HDMI anymore. Is there any procedure to reinstall all audio components?

Joshua-Riek, 2024-01-11:
There is no procedure to reinstall audio components. I would assume you accidentally broke the audio with some sort of user change. But i will double check that audio over HDMI is working on my end.

almaster76, 2024-01-11:
> There is no procedure to reinstall audio components. I would assume you accidentally broke the audio with some sort of user change. But i will double check that audio over HDMI is working on my end.

Thanks. I'll try a new clean install as soon as I can.

almaster76, 2024-01-13:
Sorry, my fault. In a new installation (via SD card) the problem does not exist. I'll have to investigate to figure out what operation of mine caused the problem. Thanks anyway.

closed 2024-01-13 by almaster76