Joshua-Riek/ubuntu-rockchip #563: Rock 5b Wifi? A2-AP6256 supported, reco for better alternative?

bagong, 2024-01-12:
Hi,

I got a Radxa Rock 5b rev 1.42 today with an wifi/bt card labled Rock Pi Wireless Module A2 - AP6256, with a BCN43456 Chip (according to radxa wiki). I first installed the "official" Debian image (kernel 5.10.110-28), and both bt and wifi were recognised, although the wifi wasn't actually able to make a connection (the wifi signal looked very low and there was no antenna provided). Bt seemed to work fine however. I then proceeded to Ubuntu Rockchip, and the wifi/bt-card was not recognized.

I have a mix of questions:
- is there a straightforward way to get the card supported in U-R? Overlay or a kernel module I could compile in this environment?
- the official case is sexy but it doesn't provide any antenna. Otoh Raspberries don't have antennas either do they? Do I need an antenna, and if yes, what should I pay attention to?
- or should I maybe buy another card, e.t. Intel AX210? They're much faster, it seems. Or are there cards that don't require an antenna?

Thanks for any input!
Best
.r.

Joshua-Riek, 2024-01-12:
It may be better to get an AX210, but could you put dmesg into a text file and upload it here? I wonder if the firmware file is just missing.

bagong, 2024-01-12:
I hope I did the right thing ;-)

bagong, 2024-01-15:
The firmware update you posted yesterday hasn't fixed my issue. I am getting an intel card this afternoon... Probably means I won't be able to verify the issue later, as disassembling these devices with passive cooling isn't really something one wants to do frequently...

bagong, 2024-01-15:
I've since switched to the Intel AX210, and there is a similar finding:
- on the vendor kernel both wifi and bluetooth work (I use a case internal paper antenna, and it works fine, I am getting 300Mbit speeds...
````
radxa@rock-5b:~$ sudo dmesg | grep wifi
[sudo] password for radxa: 
[    8.080741] wifi_disable: no parameters, enabled
[    8.080799] reg-fixed-voltage wifi-diable-gpio-regulator: wifi_disable supplying 0uV
[    9.672384] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6256
[    9.672388] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[    9.672392] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    9.672437] [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[    9.672446] [WLAN_RFKILL]: rfkill_set_wifi_bt_power: 1
[   13.818055] iwlwifi 0002:21:00.0: enabling device (0000 -> 0002)
[   13.941936] iwlwifi 0002:21:00.0: api flags index 2 larger than supported by driver
[   13.941966] iwlwifi 0002:21:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 93.8.63.28
[   13.942390] iwlwifi 0002:21:00.0: loaded firmware version 59.601f3a66.0 ty-a0-gf-a0-59.ucode op_mode iwlmvm
[   14.072613] iwlwifi 0002:21:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[   14.343776] iwlwifi 0002:21:00.0: base HW address: 90:09:df:e8:51:07
[   14.557668] Modules linked in: vfat fat zstd zstd_compress algif_skcipher zram iwlmvm hmac hci_uart sha512_generic btusb iwlwifi btqca md4 btrtl algif_hash btbcm btintel bluetooth af_alg snd_soc_simple_card snd_soc_simple_card_utils pwm_fan snd_soc_dummy_codec snd_soc_es8316 ecdh_generic bifrost_kbase ecc snd_soc_hdmi_codec snd_soc_rockchip_multicodecs snd_soc_rockchip_hdmi snd_soc_rockchip_i2s_tdm snd_soc_rockchip_spdif snd_soc_core snd_pcm_dmaengine snd_pcm snd_timer snd soundcore ledtrig_timer ledtrig_pattern ledtrig_heartbeat ledtrig_default_on fuse ip_tables x_tables ipv6 r8125(O) sdhci_of_dwcmshc dw_hdmi_qp_cec dw_hdmi_qp_i2s_audio
````

Will post from other system in a second...

Joshua-Riek, 2024-01-15:
Enter the command below, this is a bug with ubuntu:
```
sudo mv /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm  /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.bak
```

https://askubuntu.com/questions/1360175/intel-wifi-6-ax210-wifi-not-working-after-update

bagong, 2024-01-15:
Oh, wow, instant fix! Thank you, I thought this must be a similar problem, so didn't search myself. Sorry! From my side this issue can be closed. Thanks again!

Joshua-Riek, 2024-01-22:
I have been working on adding support for the Radxa Zero 3, which uses AP6212 for WiFi. During this process, I enabled drivers for broadcom WiFi devices. So, the AP6256 may work when I push the next kernel update.

bagong, 2024-01-23:
Thanks. I won't be able however to confirm this for the time being as it's too much hassle to un- and reassemble the unit, and I am happy with the AX210...

Joshua-Riek, 2024-01-23:
Gotcha, I'll close this for now then. Maybe open in the future if someone reports an issue.

defencedog, 2024-08-09:
In OrangePi 3b Wifi/BT is provided by same chip `AP6256` ... wifi works but BT doesn't. I have tested latest Joshua releases