Joshua-Riek/ubuntu-rockchip #564: Mixtile Core3588e HDMI CEC not hooked up?

StijnKuipers, 2024-01-15:
Not sure if this should be reported in linux-rockchip or here - but here goes:

I recently noticed that on ubuntu-rockchip - 4k display modes would not work on the mixtile core3588 board

comparing the dts files I found this difference:

Joshua-Riek version: 
&hdmi0 {
	status = "okay";
};

Mixtile version:
&hdmi0 {
	pinctrl-names = "default";
        pinctrl-0 = <&hdmim0_tx0_cec &hdmim1_tx0_hpd &hdmim0_tx0_scl &hdmim0_tx0_sda>;
	enable-gpios = <&gpio0 RK_PC4 GPIO_ACTIVE_HIGH>;
	status = "okay";
};

(see here -> https://github.com/mixtile-rockchip/kernel/blob/mixtile/core3588e/ubuntu/kernel5.10/arch/arm64/boot/dts/rockchip/rk3588-mixtile-core3588e.dts. ) 

Could this be the reason the 4k screen wont list all its modes? (same board runs 4k fine with the mixtile provided build)

Joshua-Riek, 2024-01-15:
Edid info is broken on the board, that is why 4k does not work. Mixtiles fix includes a hdmi driver patch that will conflict with other boards.

You could try adding the changes to the hdmi0 node then compile the kernel.

StijnKuipers, 2024-01-15:
Thanks for the quick response - will give it a try! :-)

Joshua-Riek, 2024-01-15:
I honestly want to re-write the whole device tree. I noticed multiple issues with it, and since I did a lot of work on the RK1, it's very similar.

The board would be perfect if Mixtile had the recovery pin as maskrom and not sdarc. Really limits the usability, particularly with mainline u-boot, so the system can boot from NVMe and other devices.

closed 2025-02-17 by github-actions[bot]