Joshua-Riek/ubuntu-rockchip #565: Widevine not supported

freezbeeee, 2024-01-16:
Hello, I have see a video on youtube from leepspvideo and your ubuntu support Widevine on opi 5, on my opi 5 plus, I can't go on netflix, disney+, etc... hope you can fix that.

Joshua-Riek, 2024-01-16:
It does work last I checked, does it not work for you?

freezbeeee, 2024-01-17:
Do you need some files like logs ?

Joshua-Riek, 2024-01-17:
You will need to spoof the browser, use the below command.
```
chromium-browser --user-agent="Mozilla/5.0 (X11; CrOS aarch64 15236.80.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/110.0.0.0 Safari/537.36"
```

freezbeeee, 2024-01-18:
damm that’s worked, big thanks for your job

closed 2024-01-18 by freezbeeee

bryce-g, 2024-06-13:
Hello Josh, the spoof isn't working for Netflix and other widevine dependant streaming services. Any pro tips?

nrpetonr, 2024-07-31:
confirm that browser spoofing does not work anymore.