Joshua-Riek/ubuntu-rockchip #570: Orange Pi 5 Mainline 6.7 SSH login incorrect

aidapsibr, 2024-01-19:
I'm attempting to test out the 6.7 mainline kernel, but I can't seem to login over SSH. I've confirmed with multiple nvme SSD. SSH agent is running, IP acquired, waited several minutes for user provisioning and ubuntu/ubuntu doesn't auth.

Latest release works fine, just not mainline build.

### Build tested
https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7515530857

closed 2024-01-19 by aidapsibr

Joshua-Riek, 2024-01-19:
It may take a few minutes for cloud-init to create the default user account. Please wait 1-2 minutes after the first boot and it should work just fine.

aidapsibr, 2024-01-19:
This one was user error, apologies. I accidentally flashed the official image and didn't catch my error.

aidapsibr, 2024-01-19:
I had a SM991 NVMe overheat and fry itself during the first flashing and I got a little flustered. heh

Joshua-Riek, 2024-01-19:
Ah gotcha! I'm glad its working for ya!