Joshua-Riek/ubuntu-rockchip #576: how to patch kernel for rock 5b?

ohyeah521, 2024-01-24:
Hello, I want to modify the configuration information of rock5b's kernel. How can I modify the code to patch it?

Joshua-Riek, 2024-01-28:
You can compile the kernel as shown on the wiki.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Kernel

ohyeah521, 2024-01-29:
@Joshua-Riek  Thank you for your answer. I would like to know how to modify the kernel configuration information through workflows. Is it to modify the /config/kernels/bsp.conf file?

Joshua-Riek, 2024-01-29:
Yes, you may need to remove the launchpad ppa or put your kernel package on hold so my updates do not overwrite your changes.

ohyeah521, 2024-01-29:
@Joshua-Riek  thanks

ohyeah521, 2024-01-29:
Is the following writing correct?
```
KERNEL_REPO=https://github.com/Joshua-Riek/linux-rockchip.git
KERNEL_BRANCH=linux-5.10-gen-rkr6
KERNEL_VERSION=5.10.160
KERNEL_CLONE_DIR=linux-rockchip
KERNEL_DEFCONFIG=rockchip_linux_defconfig
DPKG_BUILDPACKAGE=y
CONFIG_KPROBES=y
CONFIG_KRETPROBES=y
CONFIG_IKHEADERS=y
```

Joshua-Riek, 2024-01-29:
Yes, I looked at your build and you need to rebase the ubuntu-rockchip repo. I was doing a test build as grub was just added as a ubiquity dependency. I force pushed over the commit reverting the change for now.

ohyeah521, 2024-01-29:
@Joshua-Riek  Hello, I have encountered a new problem. I found in the .github/workflows/build.yml file.
```
run: sudo ./build.sh --board=orangepi-5 --kernel-only 
run: sudo ./build.sh --board=orangepi-5 - -rootfs-only
```
the default compiler is the orangepi-5 development board. Is this universal? Or do I need to modify it to my target development board, such as rock-5b?

Joshua-Riek, 2024-01-30:
It does not matter, currently its a hack just to build the rootfs. After the rootfs is built it will be configured per board.

closed 2024-02-19 by ohyeah521